dusty-nv/jetson-inference #1138: CUDA initialization failure while building

senturank, 2021-07-01:
[image]

We are getting a CUDA initialization failure while trying to build the package. I am using a Jetson Nano production SOM and have installed all the required packages including CUDA using the SDK manager. I even tried uninstalling & reinstalling CUDA, flashing the SD card again, but the issue persists. We did not face this issue when we were testing out on the Nano Devkit though.

Any help will be highly appreciated.

dusty-nv, 2021-07-01:
Hi @senturank, do you see the CUDA toolkit under `/usr/local/cuda` ?  Can you build and run the `deviceQuery` sample?

senturank, 2021-07-02:
Hi @dusty-nv ,
I do see the CUDA toolkit under /usr/local/cuda. But the Device query is showing errors.
[image]

I have also attached a screenshot of our nvcc version. Not able to figure our what I am doing wrong.
[image]

dusty-nv, 2021-07-02:
It seems that your CUDA driver / toolkit got it a bad state somehow, perhaps it was inadvertently upgraded.   I would backup your work and reflash the device.

closed 2023-04-27 by dusty-nv