dusty-nv/jetson-inference #1139: Loss value = Nan for CPU running and for latest PyTorch 1.9.0 and torchvision 0.10.0

sas9624, 2021-07-05:
Hello Dusty,

There are 2 case scenarios that the loss values are all Nan and I would like to ask if you have any suggestions on this matter or the solutions on how to fix it.
 
1. I tried running the Pytorch-SSD retraining on the CPU by giving the flag --use-cuda=False for the python script "train_ssd.py"
It gave Nan as a result for all typed of losses.  

2. When the Pytorch and torchvision are upgraded to PyTorch 1.9.0 and torchvision 0.10.0. No value of losses is computed. 

Now I am using 
Jetson AGX Xavier
JetPack 4.5.1
torch version 1.8.0
torchvision version 0.9.0a0+01dfa8e

dusty-nv, 2021-07-06:
Hi @sas9624, I haven't tested with PyTorch 1.9 yet or CPU-only.  In the future I will have to test PyTorch 1.9 with pytorch-ssd.

sas9624, 2021-07-06:
Hello @dusty-nv! What do you suggest to edit in the code to make it work on the CPU, do you have any ideas why the loss became Nan?

sas9624, 2021-07-08:
[image]
Hello @dusty-nv, I would like to ask if you know the reasons why the confidence value became Nan after the first iteration of CPU training. 

I re-trained the SSD model with the fruit images as in the tutorial.
In [train_ssd.py](https://github.com/dusty-nv/pytorch-ssd/blob/8ed842a408f8c4a8812f430cf8063e0b93a56803/train_ssd.py), I tried printing out the confidence value after line 120: confidence, locations = net(images)

As reported in the screenshot, the first iteration gave correctly the confidence value, then it became Nan afterward.

dusty-nv, 2021-07-15:
I found the PyTorch 1.9 issue was a problem in the wheel, with fixed PyTorch 1.9 wheel it no longer makes NaN losses (on GPU - have not tried CPU-only)

https://forums.developer.nvidia.com/t/pytorch-for-jetson-version-1-9-0-now-available/72048/876?u=dusty_nv

sas9624, 2021-07-21:
Thank you Dusty for the update!, please let me know if you have any updates about the CPU too.

closed 2023-04-27 by dusty-nv