dusty-nv/jetson-inference #1140: annotate own data

syliabe, 2021-07-06:
Hello,

Please, is there a tool to make automatic annotations of my own data, because manual annotation takes a lot of time?
I tried with cvat.org,  on the other hand it cannot annotate all the existing object in an image

thank you

dusty-nv, 2021-07-06:
Hi @syliabe, CVAT is more comprehensive than I could provide and support myself.  What do you mean when you say it can't annotate all the objects?

syliabe, 2021-07-06:
I mean in an image if I have 4 person to annotate,  the CVAT annotate just 2 instead of 4 person

dusty-nv, 2021-07-06:
I have annotated many objects per image with CVAT, more than 4.  Are you able to add more bounding boxes?

syliabe, 2021-07-06:
[image]

on my side I select automatic annotation as shown in the image above
then i downloaded it as long as pascal voc data,
we can do manual annotation but I want to avoid it as much as possible

closed 2023-04-27 by dusty-nv