dusty-nv/jetson-inference #1143: Running Tensorrt detectnet file in c++

maazullah96, 2021-07-07:
I am trying to run detect net tensorrt file trained using transfer learning toolkit . I am able to run inference on python using tensorrt and pycuda.but I have to do inference in c++ .I am facing the issue of running detect net  tensorrt in c++.I am trying to run detectnet .cpp file in ubuntu but I could not run it properly If there is any simple way to run detectnet with trt file using c++ let me know about it.
Thanks
[image]

dusty-nv, 2021-07-07:
Hi @maazullah96, I don't have support for the TLT models in this repo.  You can use them through DeepStream or here:  https://github.com/NVIDIA-AI-IOT/tlt-triton-apps

Also, to build your own C++ program that links to jetson-inference/jetson-utils, see the example CMakeLists.txt from here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/imagenet-example-2.md

maazullah96, 2021-07-08:
Thanks, @dusty-nv  for your reply.
I am trying to run the Tensorrt file with c++ . Is that possible to run jetson-inference on a PC or it is specific only for jetson? I have built the whole project using CMake and the project build completely

I was thinking that if we can run tensorrt  file on python without deepstream which is done on this link and working perfectly https://forums.developer.nvidia.com/t/how-to-run-detectnet-v2-resnet18-trt-without-deepstream/128948/12.

I just want to know is there any similar way to do run tensorrt file in c++ without deepstream.

dusty-nv, 2021-07-08:
jetson-inference is only intended for / support on Jetson, but some folks have built it on PC.

TensorRT has a C++ API also, you could take that Python code from the forum post and port it to use the C++ API.  There are C++ samples under `/usr/src/tensorrt/samples` and docs online.

closed 2023-04-27 by dusty-nv