dusty-nv/jetson-inference #1144: loading two models in the same file

mabdelbari, 2021-07-08:
Hello @dusty-nv 
I am working on a project in which I wanna use an object detection model with segmentation.
After making some customizations on both detectnet and segnet, I wanna combine together as they both use the same input stream ( live stream ).

Is it possible? 

Thank you

dusty-nv, 2021-07-08:
Yes, you just need to "hardcode" the arguments to load the models as opposed to using jetson-inference's built-in command-line parser, because that parser assumes only 1 model.

mabdelbari, 2021-07-09:
Can you provide me more details about it? 
because I think I had made something like that but it didn't work is that I had created a new argument with a different name to 
receive the segmentation network but it didn't work.

I dunno if you did mean that or not

dusty-nv, 2021-07-09:
Are you using Python or C++?

In python, it would be like:  

``` python
detectnet = jetson.inference.detectNet("ssd-mobilenet-v2")
segnet = jetson.inference.segNet("fcn-resnet18-cityscapes-512x256")
```

Or if you have custom model to load, you can do it like:

``` python
detectnet = jetson.inference.detectNet(argv=['--model=your_model/ssd_mobilenet.onnx', '--labels=your_model/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'])
```

mabdelbari, 2021-07-10:
Alright.. I will try it and keep you updated 😄 
 
Really I am grateful for you 
Many thanks 
God bless you

closed 2023-04-27 by dusty-nv