dusty-nv/jetson-inference #1146: Double bounding boxes for the same detected object in a live camera feed

Dreadlocke1984, 2021-07-10:
Hi,
I have retrained the SSD-MobileNet model on my own custom dataset (toy motorcycles). For certain orientations of the toy, I am getting two bounding boxes, sometimes with different confidence level. Is it due to limited training data? Or is the reason something else?
[image]

dusty-nv, 2021-07-10:
Try decreasing the clustering threshold here:

https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/detectNet.h#L477

Then re-run make and 'sudo make install'

[Screenshot from 2021-07-10 20-46-29]<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F67857441%2F125168044-1ad47100-e1c1-11eb-93fb-1624f3ff56a7.png&data=04%7C01%7Cdustinf%40nvidia.com%7C61fd43413949470f192508d943b7211c%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C637615276081510479%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=RWYGdHyUKdP141DCk3EpTDq14kmDH5Xe%2FGdUfGHv5%2BI%3D&reserved=0>

Dreadlocke1984, 2021-07-10:
Hi @dusty-nv. Thanks for the reply. Do I need to re-run make inside the docker container? Also will I need to create a makefile? I am completely new to all of this.

dusty-nv, 2021-07-12:
First I would test your change outside of Docker, by building the project from source.  If your change works, you can rebuild the docker container like here:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-docker.md#building-the-container

Note that rebuilding the docker container will take awhile the first time you do it, and it will also delete your `jetson-inference/python/training/detection/ssd/models` directory and `jetson-inference/python/training/detection/ssd/data` directory (so be sure to back those up if you stored your models/dataset there)

Dreadlocke1984, 2021-07-17:
Hi @dusty-nv.I built the project from source and now I am able to change the cluster threshold and rebuild the project. It is running fine. Thanks for the help. I was using the pre-built docker image before and was not aware about the building from source option. But now I better understand the difference between the two approaches. Thanks again for the help.

closed 2021-07-17 by Dreadlocke1984

dusty-nv, 2021-07-17:
You can re-build the docker container with docker/build.sh, but it will take awhile to build and also backup your models and datasets under python/training/classification and python/training/detection/ssd because those get deleted during the container build.

Closed #1146<https://github.com/dusty-nv/jetson-inference/issues/1146>.