dusty-nv/jetson-inference #1148: imageNet.classify() encountered an error classifying the image

surefyyq, 2021-07-14:
Hi @dusty-nv ,

I followed  _Re-training on the Cat/Dog Dataset_  and have no problems. In terminal, I run command line
`imagenet --model=MODEL_PATH/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=LABEL_PATH/labels.txt IMG_PATH/dog/04.jpg dog.jpg`
and it works normally.

According to `my-recognition.py` and _Coding Your Own Image Recognition Program (Python)_, I write python code like below:
```
onnx = 'MODEL_PATH/resnet18.onnx'
labels = 'Label_PATH/labels.txt'
argv = ['--model={}'.format(onnx), '--labels={}'.format(labels), '--input-blob=input_0', '--output_blob=output_0']
model_cls = jetson.inference.imageNet('resnet-18', argv)

color_img = 'IMG_PATH/dog/04.jpg'
cuda_img = jetson.utils.loadImage(color_img)

class_id, confidence = model_cls.Classify(cuda_img)
```
The [TRT] output in terminal indicates it loads model and label successfully, but finally it throws out error:
```
Traceback (most recent call last):
  File "my_code.py", line 101, in <module>
    class_id, confidence = model_cls.Classify(cuda_img)
Exception: jetson.inference -- imageNet.Classify() encountered an error classifying the image
```

Please, what may be the problem and how to handle it? 
Or, what's the suggested way to load a imageNet with named network type and model path?

dusty-nv, 2021-07-14:
Hi @surefyyq, try it like this:

``` python
model_cls = jetson.inference.imageNet(argv=['--model=MODEL_PATH/resnet18.onnx', '--labels=Label_PATH/labels.txt', '--input-blob=input_0', '--output-blob=output_0'])
```

surefyyq, 2021-07-15:
Thanks for your reply @dusty-nv ! But it still ouputs error like below:

```
[image] loaded 'IMG_PATH/dog/04.jpg'  (500x375, 3 channels)
cuda_img:  (375, 500, 3) <class 'jetson.utils.cudaImage'> <cudaImage object>
   -- ptr:      0x101070000
   -- size:     562500
   -- width:    500
   -- height:   375
   -- channels: 3
   -- format:   rgb8
   -- mapped:   true
   -- freeOnDelete: true

classification argv:  ['--model=MODEL_PATH/resnet18.onnx', '--labels=LABEL_PATH/labels.txt', '--input-blob=input_0', '--output_blob=output_0'] <class 'list'>
jetson.inference -- imageNet loading network using argv command line params

imageNet -- loading classification network model from:
         -- prototxt     (null)
         -- model        MODEL_PATH/resnet18.onnx
         -- class_labels LABEL_PATH/labels.txt
         -- input_blob   'input_0'
         -- output_blob  'output_0'
         -- batch_size   1

[TRT]    TensorRT version 7.1.3
[TRT]    loading NVIDIA plugins...
[TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
[TRT]    Registered plugin creator - ::NMS_TRT version 1
[TRT]    Registered plugin creator - ::Reorg_TRT version 1
[TRT]    Registered plugin creator - ::Region_TRT version 1
[TRT]    Registered plugin creator - ::Clip_TRT version 1
[TRT]    Registered plugin creator - ::LReLU_TRT version 1
[TRT]    Registered plugin creator - ::PriorBox_TRT version 1
[TRT]    Registered plugin creator - ::Normalize_TRT version 1
[TRT]    Registered plugin creator - ::RPROI_TRT version 1
[TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
[TRT]    Could not register plugin creator -  ::FlattenConcat_TRT version 1
[TRT]    Registered plugin creator - ::CropAndResize version 1
[TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
[TRT]    Registered plugin creator - ::Proposal version 1
[TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
[TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
[TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
[TRT]    Registered plugin creator - ::Split version 1
[TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
[TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
[TRT]    detected model format - ONNX  (extension '.onnx')
[TRT]    desired precision specified for GPU: FASTEST
[TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
[TRT]    native precisions detected for GPU:  FP32, FP16
[TRT]    selecting fastest native precision for GPU:  FP16
[TRT]    attempting to open engine cache file /home/nvidia/jetson-inference/python/training/classification/models/cat_dog/resnet18.onnx.1.1.7103.GPU.FP16.engine
[TRT]    loading network plan from engine cache... /home/nvidia/jetson-inference/python/training/classification/models/cat_dog/resnet18.onnx.1.1.7103.GPU.FP16.engine
[TRT]    device GPU, loaded /home/nvidia/jetson-inference/python/training/classification/models/cat_dog/resnet18.onnx
[TRT]    Deserialize required 3191513 microseconds.
[TRT]    
[TRT]    CUDA engine context initialized on device GPU:
[TRT]       -- layers       29
[TRT]       -- maxBatchSize 1
[TRT]       -- workspace    0
[TRT]       -- deviceMemory 29827072
[TRT]       -- bindings     2
[TRT]       binding 0
                -- index   0
                -- name    'input_0'
                -- type    FP32
                -- in/out  INPUT
                -- # dims  4
                -- dim #0  1 (SPATIAL)
                -- dim #1  3 (SPATIAL)
                -- dim #2  224 (SPATIAL)
                -- dim #3  224 (SPATIAL)
[TRT]       binding 1
                -- index   1
                -- name    'output_0'
                -- type    FP32
                -- in/out  OUTPUT
                -- # dims  2
                -- dim #0  1 (SPATIAL)
                -- dim #1  2 (SPATIAL)
[TRT]    
[TRT]    binding to input 0 input_0  binding index:  0
[TRT]    binding to input 0 input_0  dims (b=1 c=3 h=224 w=224) size=602112
[TRT]    binding to output 0 output_0  binding index:  1
[TRT]    binding to output 0 output_0  dims (b=1 c=2 h=1 w=1) size=8
[TRT]    
[TRT]    device GPU, MODEL_PATH/resnet18.onnx initialized.
[TRT]    imageNet -- loaded 2 class info entries
[TRT]    imageNet -- MODEL_PATH/resnet18.onnx initialized.
model_cls:  <jetson.inference.imageNet object at 0x7f867548d0>

Traceback (most recent call last):
  File "my_code.py", line 98, in <module>
    class_id, confidence = model_cls.Classify(cuda_img)
Exception: jetson.inference -- imageNet.Classify() encountered an error classifying the image
```

dusty-nv, 2021-07-15:
Hmm I am not sure - since imagenet.py is very similar, can you make a similar modification to it or copy it's source for your needs?

surefyyq, 2021-07-15:
Many appreciation for your quick answer @dusty-nv !

I don't make any modification , just copy imagenet.py source code to my custom file folder and rename it as my_imagenet.py. 

At this custom file path, when I run command line in terminal:
`imagenet --model=MODEL_PATH/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=LABEL_PATH/labels.txt IMG_PATH/dog/04.jpg cls_test.jpg`
It ends without error and the cls_test.jpg saved successfully.

But when I run this in terminal:
`python my_imagenet.py --model=MODEL_PATH/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=LABEL_PATH/labels.txt IMG_PATH/dog/04.jpg cls_test.jpg`
It still goes error:
```
Traceback (most recent call last):
  File "my_imagnet.py", line 47, in <module>
    class_id, confidence = model_cls.Classify(img)
Exception: jetson.inference -- imageNet.Classify() encountered an error classifying the image
```

I'm sooooo confused. How could this happen?

dusty-nv, 2021-07-15:
Do you get the same problem running `imagenet.py` without any modification?

surefyyq, 2021-07-15:
Yes...
Just for sure, the  `imagenet.py` is at `jetson-inference/python/examples/imagenet.py`, right?
And I run:
`python imagenet.py --model=MODEL_PATH/resnet18.onnx --input_blob=input_0 --output_blob=output_0 --labels=LABEL_PATH/labels.txt IMG_PATH/dog/04.jpg cls_test.jpg`
Still the same problem.

Should I try its default setting which uses googlenet?

dusty-nv, 2021-07-15:
Yes, next please try imagenet.py with the default googlenet instead of your custom model - do you get the same error?

surefyyq, 2021-07-16:
Hi @dusty-nv , sorry for the late.

I try the imagenet.py with the default googlenet and it works fine, the camera onboard is called and [TRT] reports Timing.

Is that means something is wrong with my custom model? But I trained it following  the _Re-training on the Cat/Dog Dataset_ and works as excepted when used by command line.

I will train the custom model again and try it after, wish me luck.

surefyyq, 2021-07-16:
@dusty-nv good news! It works all right after I retrain the model...

Still don't know why, but thanks very much for your time and attention!

closed 2021-07-16 by surefyyq