dusty-nv/jetson-inference #1149: Easiest way to convert inference resolution arbitrarily for SSD-Mobilenet-v2?

rocketsfallonrocketfalls, 2021-07-14:
Title.

dusty-nv, 2021-07-14:
You need to re-train the model and make modifications to the pytorch_ssd source code.  See this post:

https://forums.developer.nvidia.com/t/how-train-jetson-inference-ssd512-model/168510/6?u=dusty_nv

rocketsfallonrocketfalls, 2021-07-14:
Hey @dusty-nv, thanks for the lightning-fast answer!

I've already read that post in depth, but I am still confused. How will changing the input resolution affect the priors? 

What considerations that I should make defining the priors based in an input size, say I want to use `832x832`?

dusty-nv, 2021-07-14:
Also see https://github.com/dusty-nv/jetson-inference/issues/931#issuecomment-846808434

dusty-nv, 2021-07-14:
> What considerations that I should make defining the priors based in an input size, say I want to use `832x832`?

See the script from https://github.com/dusty-nv/jetson-inference/issues/931#issuecomment-846808434 for generating priors for other sizes

rocketsfallonrocketfalls, 2021-07-14:
Thanks a lot, @dusty-nv!

closed 2021-07-14 by rocketsfallonrocketfalls