dusty-nv/jetson-inference #1151: Required Yolo 5s in detectnet

rlrahulkanojia, 2021-07-15:
Hey @dusty-nv 

I know yolo is not supported and pre processing and post processing definition are different. I have been through all issues also.
I am ready to code all required logic but can you please help with where to start and what to change. 

Thanks
Rahul

dusty-nv, 2021-07-15:
@rlrahulkanojia here is where the pre- and post-processing is for ONNX detection model:

https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/detectNet.cpp#L728
https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/c/detectNet.cpp#L813

rlrahulkanojia, 2021-07-16:
Thanks for quick reply!

closed 2021-07-16 by rlrahulkanojia

lars-nagel, 2024-02-05:
@rlrahulkanojia Hi there! Did you succeed in coding support for YOLO? I would be also be interested in that - any hints are appreciated!