dusty-nv/jetson-inference #1152: Changing Step in SSD Training to more than 10?

eanmikale, 2021-07-15:
Training an SSD model for detection with 500k images, and there are 400k steps, and the training is going 10 steps by 10 steps. At this rate, training will take who knows how long. Is there a way to change the step to maybe 100 or more? 

Also, I could not train large datasets using the FAT32 or the Exfat standards. It might help others to know to format their external hard drive to ext4 in order to train data sets in the hundreds of thousands of images. Thank you.

dusty-nv, 2021-07-15:
Hi @eanmikale, `--debug-steps` will change how often the status is printed out, but this will not make it go faster.  You can increase the `--batch-size` if you have memory to spare.

For this very large dataset, you may be better off finding some PC or cloud instance with GPU to train your model on.  You can run the pytorch_ssd repo on an x86 machine running Ubuntu that has GPU:  https://github.com/dusty-nv/pytorch-ssd

This is the same training code, it can run on Jetson or on x86.

eanmikale, 2021-07-15:
Dusty, 

Than you. I think you are correct and a Cloud Instance is the answer. You can close this issue. Thank you. 

Regards, 

Ean Mikale, J.D.

Chief A.I. Architect

Infinite 8 Institute, L3C

***@***.***

http://www.infinite8institute.com

---- On Thu, 15 Jul 2021 12:16:05 -0500 Dustin Franklin ***@***.***> wrote ----

Hi https://github.com/eanmikale, --debug-steps will change how often the status is printed out, but this will not make it go faster.  You can increase the --batch-size if you have memory to spare.

For this very large dataset, you may be better off finding some PC or cloud instance with GPU to train your model on.  You can run the pytorch_ssd repo on an x86 machine running Ubuntu that has GPU: https://github.com/dusty-nv/pytorch-ssd

This is the same training code, it can run on Jetson or on x86.

closed 2023-04-27 by dusty-nv