dusty-nv/jetson-inference #1156: Jetson Inference Video Source Latency

irshadkn, 2021-07-19:
@dusty-nv
import jetson.utils

input = jetson.utils.videoSource("rtsp://@192.168.1.61:554/Streaming/channels/102")
output = jetson.utils.videoOutput("")

while output.IsStreaming():
image = input.Capture()
output.Render(image)
output.SetStatus("Video Viewer | {:d}x{:d} | {:.1f} FPS".format(image.width, image.height, output.GetFrameRate()))

This code is working for me. But there is a lag(latency). I know there is an option with gstreamer pipeline to reduce latency. How to do with this code? Otherwise, Is there any option to pass gstreamer pipeline in this code?

dusty-nv, 2021-07-19:
Can you try setting --input-rtsp-latency=500  (the default is 2000)

input = ***@***.***:554/Streaming/channels/102")

irshadkn, 2021-07-19:
$ python3 test.py 
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for 192.168.1.61
Opening in BLOCKING MODE
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 279 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 279 
[gstreamer] gstDecoder -- discovered video resolution: 640x360  (framerate 0.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h265, stream-format=(string)byte-stream, alignment=(string)au, width=(int)640, height=(int)360, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true, profile=(string)main, tier=(string)main, level=(string)2.1
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] rtspsrc location=rtsp://192.168.1.61:554/Streaming/channels/102 --input-rtsp-latency=0 latency=2000 ! queue ! rtph265depay ! h265parse ! omxh265dec ! video/x-raw ! appsink name=mysink
[gstreamer] gstDecoder -- failed to create pipeline
[gstreamer]    (syntax error)
[gstreamer] gstDecoder -- failed to create decoder for rtsp://192.168.1.61:554/Streaming/channels/102 --input-rtsp-latency=0
Traceback (most recent call last):
  File "test.py", line 5, in <module>
    input = jetson.utils.videoSource("rtsp://192.168.1.61:554/Streaming/channels/102 --input-rtsp-latency=0")
Exception: jetson.utils -- failed to create videoSource device

dusty-nv, 2021-07-19:
Try using video-viewer.py first:

python3 video-viewer.py --input-rtsp-latency=500 rtsp://IP:port

irshadkn, 2021-07-19:
Working with video-viewer.py
Could you pls help me to solve the issue?

dusty-nv, 2021-07-19:
Try it like this:

``` python
import jetson.utils

input = jetson.utils.videoSource("rtsp://@192.168.1.61:554/Streaming/channels/102", argv=['--input-rtsp-latency=500'])
output = jetson.utils.videoOutput("")

while output.IsStreaming():
image = input.Capture()
output.Render(image)
output.SetStatus("Video Viewer | {:d}x{:d} | {:.1f} FPS".format(image.width, image.height, output.GetFrameRate()))
```

irshadkn, 2021-07-19:
working. Thank you. Is it possible to render the output video on tkinter gui?

dusty-nv, 2021-07-19:
If you manually create your own tkinter gui it could be possible using [`cudaToNumpy()`](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-image.md#converting-to-numpy-arrays) to get the image in numpy array

irshadkn, 2021-07-20:
input = jetson.utils.videoSource("rtsp://192.168.1.61:554/Streaming/channels/102",argv=['--input-rtsp-latency=100','--input-width=352','--input-height=288'])
        while True:
            image = input.Capture()
            image = jetson.utils.cudaToNumpy(image)
            frame1 = Image.fromarray(image)
            frame1 = ImageTk.PhotoImage(frame1)
            self.video_display1.configure(image=frame1)
            self.video_display1.image = frame1

But, output video is flickering! (especially when there is a motion in the frame)

AK51, 2021-08-16:
I have tried 
python3 video-viewer.py --input-codec=h264 --input-rtsp-latency=500 rtsp://IP:port
but no improvement... delay is still 2-3seconds
The IP does not have any setting for rtsp...
Is there any way I can reduce the latency? Thx