dusty-nv/jetson-inference #1160: how to use serial in container?

whxhwhx, 2021-07-21:
I want to use the serial in the container . And I did not see the relevant parameters in command “docker/run --help”. How can I mount the serial port to the container?

dusty-nv, 2021-07-21:
Hi @whxhwhx, I haven't done this before myself - you might want to see if the steps from this article work:

https://www.losant.com/blog/how-to-access-serial-devices-in-docker

You can edit the [`sudo docker run`](https://github.com/dusty-nv/jetson-inference/blob/19ed62150b3e9499bad2ed6be1960dd38002bb7d/docker/run.sh#L168) command inside the `docker/run.sh` script as needed.

whxhwhx, 2021-07-22:
I was struggling with how to edit the "run.sh" script yesterday, and then I added a command "--device /dev/ttyTHS1:/dev/ttyTHS1" according to your suggestion.
It working. 
Thank you.

closed 2023-04-27 by dusty-nv