dusty-nv/jetson-inference #1161: How to check gstreamer changed state from READY to PAUSED using python api?

NISHANTSHRIVASTAV, 2021-07-23:
Hello All,

For reading live input stream we are using **jetson.utils.videoSource** python API.

_input = jetson.utils.videoSource('/dev/video2', argv=sys.argv)
img = input.Capture()_

But gstreamer pipeline get PAUSED the program execution is getting stuck at img=input.capture() line in code.

We also applied this logic to check whether streaming is working or not using the below code snippet:
 
_if input.IsStreaming():
      img = input.Capture()_
       
But input.IsStreaming() returns False when the streaming is open. Is this IsStreaming() function working properly?

So, How to check whether the state of the GStreamer pipeline using python API?

dusty-nv, 2021-07-23:
`input.IsStreaming()` will only return true after you call `input.Capture()` once or call `input.Open()`

`input.Capture()` will implicitly call `input.Open()` for you if the stream hasn't already been opened.

NISHANTSHRIVASTAV, 2021-08-04:
Thanks, it worked after giving a passing timeout argument i.e input.Capture(timeout=10000)

closed 2021-08-04 by NISHANTSHRIVASTAV