dusty-nv/jetson-inference #1162: output binding stuck for docker

thasin96, 2021-07-23:
I am new to jetson device inference. I have created docker image which performs object detection using tensorrt on jetson nano device. I am running the docker image by passing docker container image to `docker/run.sh` script. It works as I expected. 

I converted my app to daemon (service) in jetson nano for the same process. In Daemon service, it starts docker container as I expected. But stucks in output binding process. Here is the complete stacktrace. 

```
Jul 23 15:52:48 thasin-desktop sudo[16895]: pam_unix(sudo:session): session opened for user root by (uid=0)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: jetson.inference -- detectNet loading build-in network 'ssd-mobilenet-v2'
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: detectNet -- loading detection network model from:
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- model networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- input_blob 'Input'
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- output_blob 'NMS'
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- output_count 'NMS_1'
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- class_labels networks/SSD-Mobilenet-v2/ssd_coco_labels.txt
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- threshold 0.500000
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- batch_size 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] TensorRT version 7.1.3
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] loading NVIDIA plugins...
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::GridAnchor_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::NMS_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::Reorg_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::Region_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::Clip_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::LReLU_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::PriorBox_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::Normalize_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::RPROI_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::BatchedNMS_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Could not register plugin creator - ::FlattenConcat_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::CropAndResize version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::DetectionLayer_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::Proposal version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::ProposalLayer_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::PyramidROIAlign_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::ResizeNearest_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::Split version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::SpecialSlice_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Registered plugin creator - ::InstanceNormalization_TRT version 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] detected model format - UFF (extension '.uff')
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] desired precision specified for GPU: FASTEST
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] requested fasted precision for device GPU without providing valid calibrator, disabling INT8
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] native precisions detected for GPU: FP32, FP16
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] selecting fastest native precision for GPU: FP16
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] attempting to open engine cache file /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.7103.GPU.FP16.engine
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] loading network plan from engine cache... /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff.1.1.7103.GPU.FP16.engine
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] device GPU, loaded /usr/local/bin/networks/SSD-Mobilenet-v2/ssd_mobilenet_v2_coco.uff
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Using an engine plan file across different models of devices is not recommended and is likely to affect performance or even cause errors.
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] Deserialize required 4412227 microseconds.
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT]
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] CUDA engine context initialized on device GPU:
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] -- layers 117
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] -- maxBatchSize 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] -- workspace 0
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] -- deviceMemory 35449344
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] -- bindings 3
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] binding 0
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- index 0
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- name 'Input'
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- type FP32
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- in/out INPUT
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- # dims 3
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #0 3 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #1 300 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #2 300 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] binding 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- index 1
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- name 'NMS'
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- type FP32
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- in/out OUTPUT
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- # dims 3
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #0 1 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #1 100 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #2 7 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] binding 2
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- index 2
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- name 'NMS_1'
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- type FP32
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- in/out OUTPUT
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- # dims 3
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #0 1 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #1 1 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: -- dim #2 1 (SPATIAL)
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT]
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] binding to input 0 Input binding index: 0
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] binding to input 0 Input dims (b=1 c=3 h=300 w=300) size=1080000
Jul 23 15:53:05 thasin-desktop amnimo.log[16798]: [TRT] binding to output 0 NMS binding index: 1
```

Please let me know, what can I do to resolve the issue.

dusty-nv, 2021-07-23:
@thasin96 I haven't run as daemon before so I'm not sure what the potential side-effects are.  Is it running as root?

At the end of your log, normally this is pretty much the end of initialization.  I wonder if some other parts of the log are getting cut off?

Maybe it is actually not able to open the camera or video stream as daemon

thasin96, 2021-07-23:
Yeah, docker was not started in interactive mode. After disabling, it started working.

closed 2021-07-23 by thasin96