dusty-nv/jetson-inference #1164: How to Run Only Model Inference On Nano and Rest on Rasp?

Hana-Ali, 2021-07-27:
I have a MASK RCNN model which takes a frame, processes it to detect objects, and then displays them on the screen. When running without displaying the resulting frames on screen (without `cv2.imshow()`), the program takes about 31 MB RAM. With `cv2.imshow()`, however, it takes about 3 GB RAM to display the resulting output.

My question is, since I have an 8GB raspi and 2GB RAM Jetson Nano, can I run the model inference *only* on the Nano, and do the `cv2.imshow()` with the raspi, so that the processing can be smooth?

dusty-nv, 2021-07-27:
Hi @Hana-Ali, I doubt the program only takes 31MB RAM to process MASK-RCNN, because I believe the MASK-RCNN model itself is larger than that.  Perhaps you see the actual 3GB RAM usage during cv2.imshow() because it keeps your program running longer for a chance to see the actual usage.

You would need to transmit the image to the raspi somehow, but once received yes in theory it could work.

You may also want to disable ZRAM on your Nano and mount additional swap to give it a chance to work there:
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap

Another thing I do is use X11 tunneling over SSH to do cv2.imshow() on my Windows laptop.  I use MobaXterm and it has a built-in X11 server.  Its nice because it doesn't require you to explicitly transmit the frame at all, you just run cv2.imshow() like normal on your Jetson except it actually displays on your remote PC.  It works fine for smaller images (i.e. not streaming HD video - for that you want to do compressed RTP feed)

Hana-Ali, 2021-07-27:
Hey @dusty-nv, thinking about it now, you do have a point, but then I’m not sure how to check how much RAM is being used up by the program with/without `.imshow`

As for the ZRAM, I actually haven’t tried that yet, so I will try it soon hopefully and get back with results! 
Im not sure I fully understood MobaXterm and the last paragraph, but I’ll look into it - anything to make the model run faster, really.

Also, I just want to ask - since MRCNN is known to be quite heavy and slow, would you recommend I just go with some YOLO model instead, or try with ZRAM and tunneling? Afair both have very high precision, YOLO slightly lower (but also being lighter), so I’m not really sure if MRCNN is worth the hassle with a 2GB Nano.

Thank you so much!

dusty-nv, 2021-07-27:
MaskRCNN is primarily used for instance segmentation (+detection), so if you only need detection then yea I would go with YOLO or SSD-Mobilenet.

closed 2023-04-27 by dusty-nv