dusty-nv/jetson-inference #1167: Setting image resolution in Python script

leenremm, 2021-07-29:
Starting with the following code:

**main.py**

	import utils

	net, input, output = utils.initialize_camera_net()

	loop_counter = 0
	while loop_counter <= 10:

		img = input.Capture()

		detections = net.Detect(img)

		print("detected {:d} objects in image".format(len(detections)))

		for detection in detections:
			print(detection)

		net.PrintProfilerTimes()

		if not input.IsStreaming() or not output.IsStreaming():
			break

		loop_counter += 1

	print ("Script End")

**utils.py**

	import jetson.inference
	import jetson.utils
	import sys
	import os

	def initialize_camera_net(camera_id=0):

		input_URI = "csi://%d" % camera_id
		output_URI = ""

		argv = ['--model=models/ssd_model.onnx',
	                '--labels=models/labels.txt',
	                '--input-blob=input_0',
	                '--output-cvg=scores',
	                '--output-bbox=boxes',
	                '--output-width=1280',
	                '--output-height=720',
	                '--threshold=0.8']

    	net = jetson.inference.detectNet(argv=argv)
    	input = jetson.utils.videoSource(input_URI, argv=argv)
    	output = jetson.utils.videoOutput(output_URI, argv=argv)

    	return net, input, output

Camera resolutions available:

        GST_ARGUS: Available Sensor modes :
        GST_ARGUS: 3264 x 2464 FR = 21.000000 fps Duration = 47619048 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;
        GST_ARGUS: 3264 x 1848 FR = 28.000001 fps Duration = 35714284 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;
        GST_ARGUS: 1920 x 1080 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;
        GST_ARGUS: 1640 x 1232 FR = 29.999999 fps Duration = 33333334 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;
        GST_ARGUS: 1280 x 720 FR = 59.999999 fps Duration = 16666667 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;
        GST_ARGUS: 1280 x 720 FR = 120.000005 fps Duration = 8333333 ; Analog Gain range min 1.000000, max 10.625000; Exposure Range min 13000, max 683709000;

I followed your [aux-streaming docs](https://github.com/dusty-nv/jetson-inference/blob/master/docs/aux-streaming.md), but adding additional output setting camera resolution (`'--output-width=1280', '--output-height=720'`) don't seem to have an effect.

@dusty-nv - How do I configure the code to select another camera resolution / mode?

leenremm, 2021-07-29:
After finding these posts: 

- https://github.com/dusty-nv/jetson-inference/issues/427
- https://github.com/dusty-nv/jetson-utils/blob/705bdcc344cfba6927778c09d33c1fd61a80c758/camera/gstCamera.h

I have managed to set the camera resolution by changing:

`input = jetson.utils.videoSource(input_URI, argv=argv)`

to:

`input = jetson.utils.gstCamera(width, height, "0")`

Is this the correct way of doing this?

leenremm, 2021-07-29:
Camera resolution changes work for cameras modes that support 29 FPS. But if I choose camera mode 0: 3264 x 2464 (which can handle 21 FPS), then I get an error:

[image]

How can I avoid this error?

dusty-nv, 2021-07-29:
@leenremm if you are talking about IMX219 (i.e. Raspberry Pi Camera module v2), I don't believe 3264x2464 mode actually works because it is only a 2-lane CSI interface whereas 4 lanes would be needed for that.

closed 2023-04-27 by dusty-nv