dusty-nv/jetson-inference #1168: How to load uff model on jetson?

thasin96, 2021-07-29:
I have downloaded `SSD-Mobilenet-v2` fron this <a href="https://github.com/dusty-nv/jetson-inference/releases/tag/model-mirror-190618">link</a>. It has uff model files converted from tensorflow I guess, I want to create TensorRT engine from the uff model files like how you did. In your repository which script actually creates TensorRT engine from the uff model files.

dusty-nv, 2021-07-29:
Hi @thasin96, I used this tool:  https://github.com/AastaNV/TRT_object_detection

thasin96, 2021-07-29:
Could you tell me from where did you downloaded the tensorflow object detection models. I mean, did u download TF2 or TF1?

dusty-nv, 2021-07-29:
It was downloaded from the TF1 model zoo:  https://github.com/tensorflow/models/blob/master/research/object_detection/g3doc/tf1_detection_zoo.md

thasin96, 2021-07-30:
Hello @dusty-nv, I have gone through this github `https://github.com/AastaNV/TRT_object_detection`, I am able to convert tf model to uff but after the uff file is generated, I am getting `#assertionflattenConcat.cpp, 49` Aborted. What is the issue? 

If you want i can show you the complete stack trace.

dusty-nv, 2021-07-30:
Are you getting the error running the TRT_object_detection sample or detectNet?

I haven't tried other TensorFlow UFF detection models before, so my experience is limited.  I have had better luck using ONNX and PyTorch to get it working.

Tiryoh, 2021-08-23:
Hi @thasin96,
I have faced the same problem `#assertionflattenConcat.cpp, 49` while converting [the .uff file of `SSD-Mobilenet-v2`](https://nvidia.box.com/shared/static/jcdewxep8vamzm71zajcovza938lygre.gz) into TensorRT engine file.
Solved by re-creating .uff file.

https://github.com/rt-net/jnm_jupyternotebook/pull/8/commits/825d61fe84dacfb0e3bc22d002cf9ffaa6bb360b

I guess these are the related information.
* https://forums.developer.nvidia.com/t/problem-with-ssd-mobilenet-tensorrt-conversion/161284/8
* https://github.com/jkjung-avt/tensorrt_demos
    * https://github.com/jkjung-avt/tensorrt_demos/commit/2ea5c04edb5ba83c7aa8c79b1b1c41fbe92fbf7e

closed 2023-04-27 by dusty-nv