dusty-nv/jetson-inference #1171: Support multiple GPU

Source82, 2021-07-30:
Hi @dusty-nv 

Does train_ssd.py support multiple GPUs.
How can one easily make the best of multiple GPUs.
Thanks

dusty-nv, 2021-07-30:
@Source82 I don't believe it does and I don't have plans to add the support for it.

The pytorch-classification code which came from torchvision does support multi-GPU, but I haven't tested that scenario before.  https://github.com/dusty-nv/pytorch-classification/blob/393914b484fc946876e0a1fe37c971acdfb1fec6/train.py#L229

Mystique-orca, 2021-08-16:
Hi @dusty-nv 

The pytorch-detection code gets the GPU allocated with default (single) Cuda device, like this:
`DEVICE = torch.device("cuda:0" if torch.cuda.is_available() and args.use_cuda else "cpu")`

Now, I think that object detection is a use case where you do require to train large data, preferably with multiple GPUs. The current code will be useful in test environments, but when you look to scale up, it does not help. Even if there are GPUs available at disposal, it will support only one. And I am currently facing this issue.

As @Source82 said, wrapping the call to create_net in nn.DataParallel, could work out, or something like this:
`parallel_net = nn.DataParallel(default_Net, gpu_ids = [0,1,2])`

Once defined, the standard model training steps can be performed  just as you would do with a standard nn.Module object

Can you please let us know if you could add support for multi GPU training in pytorch-detection, anytime soon? Similar to what you have done with pytorch-classification using command line args. I bet there are a lot of folks in the community who would want that.

Thanks

closed 2023-04-27 by dusty-nv