dusty-nv/jetson-inference #1172: bad accurancy ,  re-trained the SSD network for people detection

syliabe, 2021-08-02:
Hello,

I re-trained the SSD network for people detection, and when I did an evaluation with eval-ssd.py I got only 41% accuracy,
batch size = 4,
epoch = 108,
Is there a way to have better accuracy in detection of people?

Thank you

dusty-nv, 2021-08-02:
How large is the dataset?  Typically the way to improve accuracy is to increase the size of the dataset and increase the number of training epochs.

syliabe, 2021-08-02:
I tested with 30,000 images 100 epochs, and 50,000 150 epochs but the mAP is still equal to 41%

syliabe, 2021-08-02:
I am using open Image data

dusty-nv, 2021-08-02:
I believe I have tried OpenImages people before and found it to be a challenging dataset - i.e. the content matter was often occluded, overlapping, or with many partial humans per image.  So it may be that 41% is not that bad considering the level of difficulty.  Or it may be the OpenImages people are not representative of your application and it doesn't work very well in practice.

Have you tried the model running on a camera to see if it detects people, or on images captured from your application with people?

syliabe, 2021-08-02:
Yes, I have already done some tests on a set of images, it detects people, not all but most; but when i do an evaluation with eval-ssd.py i only have 41%

I did the same for the cars (I ran the retraining on an openImage ~ 30000 image set), on the other hand for the evaluation with eval-ssd.py I had a good precision value ~ 73%

closed 2023-04-27 by dusty-nv