dusty-nv/jetson-inference #1174: I cannot clone project submodules through your methods.

sk8xiao, 2021-08-06:
Hi
I cannot clone the submodule by "git clone --recursive https://github.com/dusty-nv/jetson-inference.git"
also cannot clone the submodule by "git submodule update --recursive --init" after clone the main project.
How can i solve this problem?

error like this:
Cloning into '/home/yxy/jetson-inference/python/training/classification'...
fatal: unable to access 'https://github.com/dusty-nv/pytorch-classification/': gnutls_handshake() failed: The TLS connection was non-properly terminated.
fatal: clone of 'https://github.com/dusty-nv/pytorch-classification' into submodule path '/home/yxy/jetson-inference/python/training/classification' failed
Failed to clone 'python/training/classification'. Retry scheduled
Cloning into '/home/yxy/jetson-inference/python/training/detection'...

dusty-nv, 2021-08-06:
Hi @sk8xiao, I was just able to checkout the project again with `git clone --recursive https://github.com/dusty-nv/jetson-inference` 

Can you try it again?  Maybe it was a temporary network issue.

larryw-jimi, 2022-06-12:
Met the same issue, an I am a user from China, only part of the module can be downloaded

fatal: clone of 'https://github.com/dusty-nv/pytorch-segmentation' into submodule path '/home/jetson/jetson-inference/python/training/segmentation' failed
Failed to clone 'python/training/segmentation' a second time, aborting

dusty-nv, 2022-06-13:
Hi @larryw-jimi, instead of cloning with the `--recursive` flag, you could try running this step after cloning the repo:

``` bash
git submodule update --init
```

Alternatively, you can download a zip archive of the source code from the github webpage.