dusty-nv/jetson-inference #1179: gst Encoder Pipeline Full - Skipping Frames on Jetson Xavier NX with 20 FPS

NISHANTSHRIVASTAV, 2021-08-07:
Hi @dusty-nv,

We are detecting objects using the Mobilenet SSD model on Jetson Xavier NX using python APIs.

We are reading input frames with RTSP protocol using `jetson.utils.videoSource("rtsp://username:password@IP-ADDRESS:554", argv=sys.argv)`  and we are displaying the output inference stream of the Mobilnet SSD model with RTP protocol using 
`jetson.utils.videoOutput("rtp://REMOTE-IP:1234", argv=sys.argv + is_headless)`

While displaying the output inference streams after some time we are getting in logs on Jetson 

**gst Encoder Pipeline Full - Skipping Frames** 

_The input resolution we are using is 

Width - 1280
Height - 720_

**What is the reason for Gst Encoder Pipeline Full? and How to handle such a scenario using python APIs so that the program should run continuously and we get RTP stream on the REMOTE system?**

Please let us know if we could provide more information.

dusty-nv, 2021-08-08:
Hi @NISHANTSHRIVASTAV, I haven't seen this before so can only hypothesize - but my guess is it is related to the networking.  If you are using wifi, can you try an ethernet connection to see if the issue disappears?  If so, that would confirm the issue may be related to networking.

You can also try reducing the bitrate with the `--bitrate` argument, that will decrease the network traffic.  And you can try decreasing the `--input-width` and `--input-height` to see if it makes an impact.

Mystique-orca, 2021-08-24:
Hi @dusty-nv,
We tried reducing the `--bitrate`, and decreasing the `--input-width` and `--input-height`. And the whole setup is in ethernet connection. The streaming continues in the beginning without any problem for some time (~2 mins) and then we get the error `gstEncoder -- pipeline full, skipping frame (1382400 bytes)`, while using the jetson-utils Python APIs.

When we checked the source code of gstEncoder.cpp, this is where the condition occurs for this error: `https://github.com/dusty-nv/jetson-utils/blob/master/codec/gstEncoder.cpp/#L442
`
```
// encodeYUV
bool gstEncoder::encodeYUV( void* buffer, size_t size )
{
	if( !buffer || size == 0 )
		return false;
	
	// confirm the stream is open
	if( !mStreaming )
	{
		if( !Open() )
			return false;
	}

	**// check to see if data can be accepted**
	if( !mNeedData )
	{
		LogVerbose(LOG_GSTREAMER "gstEncoder -- pipeline full, skipping frame (%zu bytes)\n", size);
		return true;
	}
```

Now we are not able to exactly figure out why the data cannot be accepted in our case. What jetson does exactly here is, it accepts the RTSP stream from a network camera, where the object detection model runs on these frames. Once the bounding box and overlay is drawn, it forwards these frames to other device using RTP protocol, where it is rendered on VLC on the remote system.

Can you explain what is the root cause of this error (or the reason for this check)? As far as we understand, it is not related to the buffer issues. If we edit the source code to remove the condition check for `mNeedData `, will it cause any disruption?

dusty-nv, 2021-08-24:
Hi, I'm not sure, so I would go ahead and try removing the check to see what happens.  After you edit the source, re-run 'make && sudo make install'

Mystique-orca, 2021-08-24:
Yeah, okay @dusty-nv . We will try that.

Please share your insights when you try it.

Thanks

bastianhjaeger, 2021-12-01:
@NISHANTSHRIVASTAV  @Mystique-orca
Any updates here? We face more or less the same issue (but from the start, not even for a second). Would be great to know how you solved it?

cheng-an152, 2023-03-22:
Has this problem been solved? Is the coding pipeline not performing enough to cause blocking?

cheng-an152, 2023-03-22:
@dusty-nv Excuse me, Has this problem been solved? Is the coding pipeline not performing enough to cause blocking?

closed 2023-04-27 by dusty-nv