dusty-nv/jetson-inference #1183: posenet.py example not working

ooijenvangilbert, 2021-08-09:
Hi Dusty,

First off, thanks for all the examples, video's and code, it has helped me get a good understanding of AI and the Jetson Nano.
However, I think there are a couple of problems with the posenet.py example.

First when we don't select a network from the command line, the default one, assigned in the code, is not the right one

`parser.add_argument("--network", type=str, default="ssd-mobilenet-v2", help="pre-trained model to load (see below for options)")`

but, when I change this, to let's say densenet121-body I get another error from 

`net = jetson.inference.poseNet(opt.network, sys.argv, opt.threshold) `

[TRT] poseNet -- failed to load topology json from 'networks/Pose-ResNet18-Body/human_pose.json' 

are my models in the wrong place or is something else going wrong here because it's seems to be looking at the resent network and not the densenet. 

thanks in advance

dusty-nv, 2021-08-09:
Thanks @ooijenvangilbert - just patched this in https://github.com/dusty-nv/jetson-inference/commit/7a1f961bf1caeb1f29bcff24ba96fcb6bfba5d21

Rebuilding the containers now with this fix, and will push the updates to DockerHub.

ooijenvangilbert, 2021-08-09:
thanks for the quick patch @dusty-nv 

after a full git clone and rebuild i still get the following error when trying to load the network :

`net = jetson.inference.poseNet(opt.network, sys.argv, opt.threshold)`

poseNet -- loading pose estimation model from:
        -- model        networks/Pose-DenseNet121-Body/pose_densenet121_body.onnx
        -- topology     networks/Pose-DenseNet121-Body/human_pose.json
        -- colors       networks/Pose-DenseNet121-Body/colors.txt
        -- input_blob   'input'
        -- output_cmap  'cmap'
        -- output_paf   'paf'
        -- threshold    0.150000
        -- batch_size   1

[TRT]    poseNet -- failed to load topology json from 'networks/Pose-DenseNet121-Body/human_pose.json'
Segmentation fault (core dumped)

this happens with the resnet and the densenet networks. can you point me at what i am doing wrong?

dusty-nv, 2021-08-09:
Can you check that you have the files `jetson-inference/networks/Pose-DenseNet121-Body/human_pose.json` ?

ooijenvangilbert, 2021-08-09:
yes, the file is there but it sits in jetson-inference/data/networks/Pose-DenseNet121-Body

dusty-nv, 2021-08-09:
`jetson-inference/networks/Pose-DenseNet121-Body/human_pose.json` is the correct and expected path for it

I wonder if it's corrupted?  Can you try running the model downloader tool again to download the pose models again?

ooijenvangilbert, 2021-08-09:
I downloaded the networks with the model downloader tool. it put them in the jetson-inference/DATA/networks folder. 
but i have a shadow folder jetson-inference/networks with the same content.

the error persists.

dusty-nv, 2021-08-09:
woops sorry, you are correct in that ` jetson-inference/data/networks` is actually the correct path.  This is symlink'd to `jetson-inference/build/aarch64/bin/networks`

Are you able to run the following?

``` bash
cd jetson-inference/build/aarch64/bin
cat networks/Pose-DenseNet121-Body/human_pose.json
```

If so, posenet/posenet.py should be able to find it if you run them from `jetson-inference/build/aarch64/bin`
If you want to run them from anywhere, do a `sudo make install`

ooijenvangilbert, 2021-08-09:
hi @dusty-nv 

yes i was able to run the cat command.
i am also able to run the posenet.py from the bin directory, it's the first run so long build.

the sudo make install has been run from the build folder (as per the described steps)

will it be working now from anywhere? i am starting my jetson AI Specialist certification and had an idea with the posenet network.

dusty-nv, 2021-08-09:
> will it be working now from anywhere?

Ah sorry about that - another thing needed patched in order for this to work (see https://github.com/dusty-nv/jetson-inference/commit/f27cb53717fcdde23ad686f515fd15ca57fa3d8c)

If you do the following it should work from anywhere then:

``` bash
$ cd jetson-inference
$ git pull origin master
$ cd build
$ make 
$ sudo make install
```

ooijenvangilbert, 2021-08-09:
oh no worries @dusty-nv , it's fixed. doing a happy dance with the resnet18-body running of course :-)

thanks for you assistance

dusty-nv, 2021-08-09:
OK, great!  Thanks for working through that with me :)

ooijenvangilbert, 2021-08-09:
solved - closing.

closed 2021-08-09 by ooijenvangilbert

monophonics, 2024-02-25:
Should we run this outside of docker container? 
  
  $ cd jetson-inference
  $ git pull origin master
  $ cd build
  $ make 
  $ sudo make install

dusty-nv, 2024-02-25:
> Should we run this outside of docker container?

@monophonics that is close, see here for the build procedure if you want to use jetson-inference outside of container:  https://github.com/dusty-nv/jetson-inference/blob/master/docs/building-repo-2.md#quick-reference