dusty-nv/jetson-inference #1184: (jetson-inference) Error in training image classification model

b2992872, 2021-08-10:
I run the camera to test my model. My "val" folder has 100 training photos, and there are 20 photos in the "train" folder, but the result is 100% regardless of whether there is a target or not.

So far, the content is the same as in the example https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-collect.md

[image]
[image]

dusty-nv, 2021-08-10:
> My "val" folder has 100 training photos, and there are 20 photos in the "train" folder

Hi @b2992872, you should have that reversed - val folder should have 20 photos, while train folder should have 80 photos.  You can just move them from one folder to another by renaming the folders in the file browser or with the `mv` command.

Then train your model again.  If it still isn't working well, you probably need even more photos in train folder.

b2992872, 2021-08-11:
Now there are 20 training photos in my "val" folder and 200 photos in the "train" folder, but the result is the same. My training images contain different lights. Is there any other problem?

dusty-nv, 2021-08-11:
Oh, try exporting the ONNX model again and deleting the .engine file so that TensorRT regenerates it.

What accuracy did PyTorch train the model to?

b2992872, 2021-08-11:
This is the training result
[image]

dusty-nv, 2021-08-11:
At the beginning of training, how many classes does it find?

In your train folder, how many images are in each class sub-folder?

[118]<https://nam11.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F61401570%2F128953970-a7961ea9-0d19-4160-974d-8b6a761d5073.png&data=04%7C01%7Cdustinf%40nvidia.com%7Cde2719105b234134897608d95c63ac2b%7C43083d15727340c1b7db39efd9ccc17a%7C0%7C0%7C637642405427715000%7CUnknown%7CTWFpbGZsb3d8eyJWIjoiMC4wLjAwMDAiLCJQIjoiV2luMzIiLCJBTiI6Ik1haWwiLCJXVCI6Mn0%3D%7C1000&sdata=Zqiu373tIegqVbEgqqbbgZBe3P2j6I1w7R4TpCCBNkQ%3D&reserved=0>

b2992872, 2021-08-11:
I only have one category. There are 20 photos in the "val" folder, 200 training photos in the "train" folder, and no photos in the "test" folder.

b2992872, 2021-08-11:
After I am sure that both categories are found, everything is normal, thank you very much!

closed 2023-04-27 by dusty-nv