dusty-nv/jetson-inference #1185: how to use own detection onnx model?its to long for trt calculation

laolihaile, 2021-08-10:
It takes too long to execut the command` detectnet.py --model=../training/detection/ssd/models/fruit/ssd-mobilenet.onnx --labels=../training/detection/ssd/models/fruit/labels.txt --input-blob=input_0 --output-cvg=scores --output-bbox=boxes csi://0` inside `jetson-inference/python/examples`
I saw that it took a long time to do the TRT calculation.
How can I shorten the startup time?
How to use onnx in Python code when I add the `argument parser. Add_argument ("--model",type= STR,default(../training/detection/ssd/models/fruit/ssd-mobilenet.onnx)) `on `detectnet.py`, it doesn't work.
How do I need to change the code in detectnet.py to use my onNX model?
`net = jetson.inference.detectNet(opt.network,sys.argv,opt.threshold)`

dusty-nv, 2021-08-10:
> I saw that it took a long time to do the TRT calculation.
> How can I shorten the startup time?

It should only take a long time the first time - it caches the TRT engine to disk.  It should load the model relatively quickly from then on.

> How do I need to change the code in detectnet.py to use my onNX model?

Try it like this:

``` python
net = jetson.inference.detectNet(argv=['--model=../training/detection/ssd/models/fruit/ssd-mobilenet.onnx', '--labels=../training/detection/ssd/models/fruit/labels.txt', '--input-blob=input_0', '--output-cvg=scores', '--output-bbox=boxes'])
```

laolihaile, 2021-08-11:
> It should only take a long time the first time - it caches the TRT engine to disk. It should load the model relatively quickly from then on.
>
>
> Try it like this:
>
> ```python
> net = [...]

Thank you very much!I will try this method.
I was confused by the fact that every time I ran the same command and model, it took me a long time (minutes or so).

dusty-nv, 2021-08-11:
> I have another question that Why only changes made to `detectnet.py` in folder `jetson-inference/python/examples` take effect?Changes made when copying `detectnet.py` to another folder do not take effect?

`jetson-inference/python/examples` is the master copy.  When you run `cmake` or `make install`, they get copied to other locations like `build/aarch64/bin` and `/usr/local/bin`

closed 2021-08-16 by laolihaile