dusty-nv/jetson-inference #1186: jetson.utils.videoOutput behavior on Jetson.utils.videoInput when using RTSP stream seems broken

ooijenvangilbert, 2021-08-10:
Hi,

i have a pretty standard :
```
input = jetson.utils.videoInput('/dev/video1')
output = jetson.utils.videoOutput()
```

this works, the stream is on the display

but when I change the input stream to RTSP 
```
input = jetson.utils.videoInput('rtsp://a:b@192.168.1.1:88/videoMain')
output = jetson.utils.videoOutput()
```

I get no output to the display.

even when I force videoOutput('display://0') still no luck
when I write this to a stream, the mp4 file is ok, so everything is working.

how can de input stream source change the output stream behavior?

Thanks

dusty-nv, 2021-08-10:
> how can de input stream source change the output stream behavior?

I'm not sure, but try creating `output = jetson.utils.videoOutput()` first.

ooijenvangilbert, 2021-08-10:
Hey Dusty,

That actually fixed it. 
I can continue, thanks :-)

But, is this expected behavior? should it be at least documented on the 'camera streaming and multimedia page' ?

Thanks.

dusty-nv, 2021-08-10:
> But, is this expected behavior?

It's not expected, I'm still trying to track down the bug.  No idea why creating RTSP feed in particular messes with the OpenGL, especially when RTP works fine...

ooijenvangilbert, 2021-08-10:
It's an older foscam ip camera. 
I also tried the raspberry pi setup with tpclientsrc but that was a no go (works in openecv tho)

If I can do any testing for you or you need more info, let me know.

ooijenvangilbert, 2021-11-03:
> [#1186 (comment)](https://github.com/dusty-nv/jetson-inference/issues/1186#issue-965150247)
> Where exactly do you put these lines? Which file? I have built from source and have a custom model... Thanks

I don't understand your question, can you give a little bit more information?

closed 2023-04-27 by dusty-nv