dusty-nv/jetson-inference #1188: fatal error: connect_parts.hpp

BiCffQ, 2021-08-11:
/home/x/jetson-inference/c/poseNet.cpp:32:10: fatal error: connect_parts.hpp: 没有那个文件或目录
 #include "connect_parts.hpp"
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
CMakeFiles/jetson-inference.dir/build.make:1207: recipe for target 'CMakeFiles/jetson-inference.dir/c/poseNet.cpp.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/poseNet.cpp.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

Jason-xy, 2021-08-16:
Hi, did you solve this error？I have met the same situation today >_<

dusty-nv, 2021-08-16:
Did you clone the repo with `--recursive`?   It seems like the `jetson-inference/plugins/pose` submodule is missing

Jason-xy, 2021-08-16:
> Did you clone the repo with `--recursive`? It seems like the `jetson-inference/plugins/pose` submodule is missing
Hi Dusty,
Thank you so much for your suggestion! As you said, I successfully completed the build after re-cloning the repository. Probably due to network reasons, the submodule was not successfully cloned.  Thanks!

closed 2023-04-27 by dusty-nv