dusty-nv/jetson-inference #1190: data augmentation transforms: RandomResizedCrop not available

silent-code, 2021-08-11:
Hi Dusty, I'm trying to do data augmentation using pytorch's RandomResizedCrop. However it does not appear to be included in the ssd vision transforms used in jetson-inference. 

I was wondering why the data augmentation transforms offered in the jetson-inference torchvision [here](https://github.com/dusty-nv/pytorch-ssd/blob/master/vision/transforms/transforms.py) vs. [the original](https://github.com/pytorch/vision/blob/master/torchvision/transforms/transforms.py) are not the same?

Thanks for your help!

dusty-nv, 2021-08-11:
> I was wondering why the data augmentation transforms offered in the jetson-inference torchvision [here](https://github.com/dusty-nv/pytorch-ssd/blob/master/vision/transforms/transforms.py) vs. [the [...]

Hi @silent-code, I'm not exactly sure, if think the original upstream author wanted to use cv2 to do the transformations so they could more easily do realtime cv demo?   https://github.com/qfgaohao/pytorch-ssd

Not sure - some of them seem custom for bounding box data / ect.

silent-code, 2021-08-11:
Ah I see. Thanks for the quick response Dusty.

closed 2021-08-11 by silent-code