dusty-nv/jetson-inference #1191: Python 3.7 docker container

chang-1, 2021-08-13:
Is there a way to run jetson-inference examples inside the docker container with python 3.7 or 3.8?

dusty-nv, 2021-08-13:
I haven't attempted this, but I believe if you modify the dockerfile to install dev packages for python 3.7 and 3.8, the jetson-inference should detect it and build bindings for python 3.7 when you rebuild the docker container.

For 3.8 you at least need to add `3.8` to these parts of the CMakeLists (before building the docker container)

* https://github.com/dusty-nv/jetson-inference/blob/5601ec6fd538ad1c5a197d077234590ba90d404a/python/CMakeLists.txt#L8
* https://github.com/dusty-nv/jetson-utils/blob/1ee0494b196b488b4b95e89fa2d72366d3cf4879/python/CMakeLists.txt#L8

chang-1, 2021-08-13:
> I haven't attempted this, but I believe if you modify the dockerfile to install dev packages for python 3.7 and 3.8, the jetson-inference should detect it and build bindings for python 3.7 when you [...]

Thanks for your prompt reply, Dusty-nv.  I will absolutely test your answer and keep you posted.

closed 2023-04-27 by dusty-nv