dusty-nv/jetson-inference #1193: re-training of SSD network

syliabe, 2021-08-16:
Hello,

I opted for a re-training of SSD network, one for the detection of people and the other for the detection of the vehicles, after several tests, I noticed that the precision is always the same, no improvement of the networks knowing that I have well increase the database (openImage) and also epoch/batch-size of each network,
I did tests with my own data the precision is reduced!

Any suggestions please?

Thanks

dusty-nv, 2021-08-16:
Forum cross-post here:  https://forums.developer.nvidia.com/t/re-training-of-ssd-network/186646/

sridhatta, 2021-08-24:
Can you please elaborate on the metrics you got after training and what was the training all about(is it finetuning) and how did u increase on the size of the dataset?

closed 2023-04-27 by dusty-nv