dusty-nv/jetson-inference #1196: error: ‘DimsCHW’ in namespace ‘nvinfer1’ does not name a type

ravali27, 2021-08-18:
Hello,

I got the following error while compiling code from https://github.com/fan0210/ros_people_detect .  I am running this on jetson Xavier with jetpack 4.6 and tensorRT 8.0.1. I am new to these things can you please suggest me some ways to solve it.

Thanks a lot :)

```
iss@iss-desktop:~/catkin_ws$ catkin_make
Base path: /home/iss/catkin_ws
Source space: /home/iss/catkin_ws/src
Build space: /home/iss/catkin_ws/build
Devel space: /home/iss/catkin_ws/devel
Install space: /home/iss/catkin_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/iss/catkin_ws/build"
####
make: Warning: File 'Makefile' has modification time 214 s in the future
make: warning:  Clock skew detected.  Your build may be incomplete.
####
#### Running command: "make -j4 -l4" in "/home/iss/catkin_ws/build"
####
make: Warning: File 'Makefile' has modification time 214 s in the future
make[1]: Warning: File 'CMakeFiles/Makefile2' has modification time 215 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/_msgs_generate_messages_check_deps_BoundingBox.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/sensor_msgs_generate_messages_nodejs.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/_msgs_generate_messages_check_deps_DetectedTargets.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/std_msgs_generate_messages_nodejs.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'msgs/CMakeFiles/_msgs_generate_messages_check_deps_DetectedTargets.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/sensor_msgs_generate_messages_nodejs.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'msgs/CMakeFiles/_msgs_generate_messages_check_deps_BoundingBox.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/std_msgs_generate_messages_nodejs.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
[  0%] Built target std_msgs_generate_messages_nodejs
[  0%] Built target sensor_msgs_generate_messages_nodejs
make[2]: Warning: File 'msgs/CMakeFiles/_msgs_generate_messages_check_deps_Target.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/std_msgs_generate_messages_py.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'msgs/CMakeFiles/std_msgs_generate_messages_py.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'msgs/CMakeFiles/_msgs_generate_messages_check_deps_Target.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
[  0%] Built target std_msgs_generate_messages_py
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'msgs/CMakeFiles/sensor_msgs_generate_messages_py.dir/progress.make' has modification time 214 s in the future
[  0%] Built target _msgs_generate_messages_check_deps_BoundingBox
make[2]: Warning: File 'msgs/CMakeFiles/std_msgs_generate_messages_cpp.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
[  0%] Built target _msgs_generate_messages_check_deps_DetectedTargets
make[2]: Warning: File 'msgs/CMakeFiles/sensor_msgs_generate_messages_py.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'msgs/CMakeFiles/sensor_msgs_generate_messages_cpp.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/std_msgs_generate_messages_cpp.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
[  0%] Built target sensor_msgs_generate_messages_py
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
make[2]: Warning: File 'msgs/CMakeFiles/std_msgs_generate_messages_eus.dir/progress.make' has modification time 214 s in the future
make[2]: Warning: File 'msgs/CMakeFiles/sensor_msgs_generate_messages_cpp.dir/progress.make' has modification time 214 s in the future
make[2]: warning:  Clock skew detected.  Your build may be incomplete.
[  0%] Built target std_msgs_generate_messages_cpp
make[2]: Warning: File 'msgs/CMakeFiles/sensor_msgs_generate_messages_eus.dir/progress.make' has modification time 214 s in the future
[  0%] Built target sensor_msgs_generate_messages_cpp
[  0%] Built target _msgs_generate_messages_check_deps_Target
[... 515 lines cut ...]
                 from /home/iss/catkin_ws/src/people_detect/src/PeopleDetect.cpp:1:
/home/iss/catkin_ws/src/people_detect/src/PeopleDetect.cpp: In member function ‘const std::vector<cv::Rect_<int> >& PeopleDetect::detect(const cv::Mat&)’:
/home/iss/catkin_ws/src/people_detect/src/PeopleDetect.cpp:47:32: warning: ‘cudaError_t cudaThreadSynchronize()’ is deprecated [-Wdeprecated-declarations]
     CUDA(cudaThreadSynchronize());
                                ^
/home/iss/catkin_ws/src/people_detect/include/net/cudaUtility.h:20:36: note: in definition of macro ‘CUDA’
 #define CUDA(x)    cudaCheckError((x), #x, __FILE__, __LINE__)
                                    ^
In file included from /usr/local/cuda/include/channel_descriptor.h:61:0,
                 from /usr/local/cuda/include/cuda_runtime.h:95,
                 from /home/iss/catkin_ws/src/people_detect/include/net/cudaUtility.h:9,
                 from /home/iss/catkin_ws/src/people_detect/include/net/cudaMappedMemory.h:9,
                 from /home/iss/catkin_ws/src/people_detect/include/PeopleDetect.h:6,
                 from /home/iss/catkin_ws/src/people_detect/src/PeopleDetect.cpp:1:
/usr/local/cuda/include/cuda_runtime_api.h:957:57: note: declared here
 extern __CUDA_DEPRECATED __host__ cudaError_t CUDARTAPI cudaThreadSynchronize(void);
                                                         ^~~~~~~~~~~~~~~~~~~~~
people_detect/CMakeFiles/people_detect_node.dir/build.make:110: recipe for target 'people_detect/CMakeFiles/people_detect_node.dir/src/PeopleDetect.cpp.o' failed
make[2]: *** [people_detect/CMakeFiles/people_detect_node.dir/src/PeopleDetect.cpp.o] Error 1
CMakeFiles/Makefile2:3268: recipe for target 'people_detect/CMakeFiles/people_detect_node.dir/all' failed
make[1]: *** [people_detect/CMakeFiles/people_detect_node.dir/all] Error 2
Makefile:140: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
```

benb0jangles, 2021-09-15:
Similar error:

```
[ 35%] Building NVCC (Device) object CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o
/home/ben1/jetson-inference-python/c/tensorNet.h(41): error: namespace "nvinfer1" has no member "DimsCHW"

/home/ben1/jetson-inference-python/c/tensorNet.h(418): error: exception specification for virtual function "tensorNet::Profiler::reportLayerTime" is incompatible with that of overridden function "nvinfer1::IProfiler::reportLayerTime"

/home/ben1/jetson-inference-python/c/tensorNet.h(403): error: exception specification for virtual function "tensorNet::Logger::log" is incompatible with that of overridden function "nvinfer1::ILogger::log"

3 errors detected in the compilation of "/tmp/tmpxft_000016ae_00000000-8_detectNet.compute_72.cpp1.ii".
CMake Error at jetson-inference_generated_detectNet.cu.o.cmake:279 (message):
  Error generating file
  /home/ben1/jetson-inference-python/build/CMakeFiles/jetson-inference.dir/c/./jetson-inference_generated_detectNet.cu.o

CMakeFiles/jetson-inference.dir/build.make:63: recipe for target 'CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o' failed
make[2]: *** [CMakeFiles/jetson-inference.dir/c/jetson-inference_generated_detectNet.cu.o] Error 1
CMakeFiles/Makefile2:67: recipe for target 'CMakeFiles/jetson-inference.dir/all' failed
make[1]: *** [CMakeFiles/jetson-inference.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2
```
Jetson Nano 2GB:

NVIDIA Jetson Nano (Developer Kit Version)
 L4T 32.6.1 [ JetPack 4.6 ]
   Ubuntu 18.04.6 LTS
   Kernel Version: 4.9.253-tegra
 CUDA 10.2.300
   CUDA Architecture: 5.3
 OpenCV version: 4.1.1
   OpenCV Cuda: NO
 CUDNN: 8.2.1.32
 TensorRT: 8.0.1.6
 Vision Works: 1.6.0.501
 VPI: ii libnvvpi1 1.1.12 arm64 NVIDIA Vision Programming Interface library
 Vulcan: 1.2.70

dusty-nv, 2021-09-15:
Are you using the latest repo from master?  Can you try re-cloning?  I'm able to build on JetPack 4.6 without issues.

benb0jangles, 2021-09-15:
> Are you using the latest repo from master? Can you try re-cloning? I'm able to build on JetPack 4.6 without issues.
> […](#)

About to retry with a fresh install.

benb0jangles, 2021-09-15:
Using a fresh install on Jetson Nano 2GB
no apt update or apt upgrade

then:
git clone --recursive https://github.com/dusty-nv/jetson-inference
cd jetson-inference 
mkdir build $ cd build
cmake ../

using default model loader selection

make
sudo make install
sudo ldconfig

first test with:
cd jetson-inference/build/aarch64/bin 

run example:
./imagenet-console.py --network=googlenet images/orange_0.jpg output_0.jpg 

Everything working fine :)

Thanks.

benb0jangles, 2021-09-15:
FYI about to test some detection examples using the drone:
[image]

closed 2023-04-27 by dusty-nv