dusty-nv/jetson-inference #1197: 2688x1520 input has problem in object detection...

AK51, 2021-08-19:
Hi, when I use 1280x720 input, everything works.
However, I want to object detect some small objects, so, I tune up my IP camera to 2688 x 1520
When I did the camera capture for detection, I can only capture the first image, the camera view freezes. And I need to restart the camera capture for another capture.
For testing, I have 4 pictures in data set (mix), when I did the training, it keeps printing this error at epoch 0
OpenBLAS Warning : Detect OpenMP Loop and this application may hang. Please rebuild the library with USE_OPENMP=1 option.

May I know if 2688 x 1520 input is ok? Or there is some code I need to modify? Thx
And what is a good way to detect small objects? Thx

dusty-nv, 2021-08-19:
The input to the object detection network is 300x300, so increasing the resolution of the camera is unlikely to help.  No matter the camera resolution, it gets downscaled to 300x300 for the DNN.

However you can see these threads for a discussion about training SSD at 512x512 resolution instead of 300x300:

* https://forums.developer.nvidia.com/t/how-train-jetson-inference-ssd512-model/168510/6?u=dusty_nv
* https://github.com/dusty-nv/jetson-inference/issues/931

AK51, 2021-08-20:
Thanks for your great help.
It is working. :>

closed 2021-08-20 by AK51