dusty-nv/jetson-inference #1198: RTSP problem (gstDecoder -- Could not get/set settings from/on resource)

AK51, 2021-08-23:
Hi, 

My other IP cameras work except this unknown brand. I can view it in my window VLC player.
This command works and I can see the pop up window video
(gst-launch-1.0 uridecodebin uri=‘rtsp://RTSP_URI’ ! nvoverlaysink)
But this command does not work
video-viewer --input-codec=h264 rtsp://192.168.50.3:554/user=…&password=…&channel=1&stream=0.sdp?
There is an error of “gststreamer rtspsrc0 ERROR Could not get/set settings from/on resources”. When I google it, this error is related to gst-launch (but I have no problem with gst-launch, only the video-viewer and detectnet…)

Here is the youtube for a better explanation
https://youtu.be/LVung1DwuEg

And I have posted it in https://forums.developer.nvidia.com/t/failed-to-capture-video-frame-using-rtsp/186446/9

May I know how to solve this problem? Do I need to modify the code?

Thx

dusty-nv, 2021-08-23:
Hi @AK51, please refer to my reply on the forums here:  https://forums.developer.nvidia.com/t/failed-to-capture-video-frame-using-rtsp/186446/11?u=dusty_nv

closed 2021-08-24 by AK51

arnold-momimo, 2024-03-13:
I got the same error, but on the `udpsink` element, and not the `rtspsrc` as in yours. Your answer on the NVIDIA dev forums post about it having to do with the rtsp string led me to double-check my own rtsp uri: `rtsp://localhost:8554/...`
Suffice it to say, using the full ip in place of the alias resolved the issue.