dusty-nv/jetson-inference #1200: PoseNet - files and models not matching tutorial

NathanielCustom, 2021-08-24:
Let me preface my question(s) by saying that I have a lot of pockets in my understanding, so make no assumptions on my competence...

I have successfully followed the examples in Hello AI World using DetectNet on the Jetson Nano including retraining the SSD_MobileNet model for some of my own applications, so I know that I have at least a minimally functioning system.

I want to experiment with PoseNet. I am trying to follow the guide at https://github.com/dusty-nv/jetson-inference/blob/master/docs/posenet.md. I am running into a couple issues...

1. ) I am not convinced that any of the three models are on my system. They make no appearance in jetson-inference/data/models. They are also don't appear to be listed in the model downloader (tools/download-models.sh).

2. ) When I try to run python/examples/posenet.py I get the following error which is raised on line 33 and line 49 whenever jetson.interference.poseNet is called:
> module 'jetson.inference' has no attribute 'poseNet'

I presume that there must be something I have overlooked which would account for these two errors/symptoms... any thoughts?

For added confusion, until I pulled the latest with $ git pull origin master, there were no C or Python PoseNet files so I manually copied them into their respective folders and mounted the folders within the docker/run.sh script. It looks like there were some changes made in the last two weeks to the PoseNet files. My J.Nano being a few months old, my original pull may have suffered from some outdated files?

dusty-nv, 2021-08-24:
Hi there, I would re-clone the repo from scratch to get it back in the latest state.  If you are on JetPack 4.4.1 or newer, the containers have been updated for poseNet

NathanielCustom, 2021-08-24:
Thanks @dusty-nv.
For completeness, I reflashed the SD card with the latest Jetpack 4.6. Previously, I had flashed it with Jetpack 4.5.1.
The models are now listed in the model downloader and I am able to run posenet.py from `jetson-inference/build/aarch64/bin`. So I have a working solution.

But what is going on with needing to run the example from the `aarch64/bin` folder? This was addressed in https://github.com/dusty-nv/jetson-inference/issues/1183.

What I don't understand is the suggestion to run the 'make' command per comment:
https://github.com/dusty-nv/jetson-inference/issues/1183#issuecomment-895504910
- The `jetson-inference/build` folder is only accessible when in the docker container.
- A git pull is completed outside the container.
- There is no makefile in the `build` folder.

dusty-nv, 2021-08-24:
> But what is going on with needing to run the example from the `aarch64/bin` folder? This was addressed in #1183.

It appears I forgot to update the JP 4.6 container after this fix - try running `sudo docker pull dustynv/jetson-inference:r32.6.1` and re-running `docker/run.sh`.  Then that should be fixed.

> What I don't understand is the suggestion to run the 'make' command per comment:
> [#1183 (comment)](https://github.com/dusty-nv/jetson-inference/issues/1183#issuecomment-895504910)

In this other comment, the person was building from source and not using the container.

NathanielCustom, 2021-08-24:
Perfect! Updating the container and repulling the container worked. I can run the example (`posenet.py`) from the `python/examples` folder. The only suggestion I would make is to adjust the poseNet documentation to be explicit as to which folder users should be running the file from.
https://github.com/dusty-nv/jetson-inference/blob/master/docs/posenet.md

Thanks for the clarification on 'make'.

closed 2021-08-24 by NathanielCustom