dusty-nv/jetson-inference #1202: cmake error, Python 3.7 wasn't found

YaoJiawei329, 2021-08-25:
Hi, all. I want to run this repo, but error occurs. 
The first time I run  $cmake ../    in /jetson-inference/build, it turned out
......
-- detecting Python 3.7...
-- Python 3.7 wasn't found
......

Then, I delete 3.7 in /jetson-inference/python/CMakeLists.txt, line 8
set(PYTHON_BINDING_VERSIONS 2.7 3.6 3.7 )  ------->  set(PYTHON_BINDING_VERSIONS 2.7 3.6). 

Then, delete /jetson-inference/build, and $mkdir build, $cd build, $cmake../
still has the same error:
......
-- detecting Python 3.7...
-- Python 3.7 wasn't found
......

Then I see this [https://github.com/dusty-nv/jetson-inference/issues/899](https://github.com/dusty-nv/jetson-inference/issues/899), so I install Python3.7, $sudo apt-get install libpython3.7-dev    , then rerun $cmake ../ , still cannot find Python3.7, so I think maybe it is not about Python3.7. I check CMakeError.log, but I cannot figure out where is the problem. Here is the CMakeError.log:

Performing C SOURCE FILE Test CMAKE_HAVE_LIBC_PTHREAD failed with the following output:
Change Dir: /home/nvidia/jetson-inference/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make -f Makefile cmTC_588eb/fast && /usr/bin/make  -f CMakeFiles/cmTC_588eb.dir/build.make CMakeFiles/cmTC_588eb.dir/build
make[1]: Entering directory '/home/nvidia/jetson-inference/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_588eb.dir/src.c.o
/usr/bin/cc -DCMAKE_HAVE_LIBC_PTHREAD  -fPIC  -o CMakeFiles/cmTC_588eb.dir/src.c.o -c /home/nvidia/jetson-inference/build/CMakeFiles/CMakeTmp/src.c
Linking C executable cmTC_588eb
/home/nvidia/cmake-3.20.4-linux-aarch64/bin/cmake -E cmake_link_script CMakeFiles/cmTC_588eb.dir/link.txt --verbose=1
/usr/bin/cc -fPIC  -rdynamic CMakeFiles/cmTC_588eb.dir/src.c.o -o cmTC_588eb 
CMakeFiles/cmTC_588eb.dir/src.c.o: In function `main':
src.c:(.text+0x48): undefined reference to `pthread_create'
src.c:(.text+0x50): undefined reference to `pthread_detach'
src.c:(.text+0x58): undefined reference to `pthread_cancel'
src.c:(.text+0x64): undefined reference to `pthread_join'
src.c:(.text+0x74): undefined reference to `pthread_atfork'
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_588eb.dir/build.make:98: recipe for target 'cmTC_588eb' failed
make[1]: *** [cmTC_588eb] Error 1
make[1]: Leaving directory '/home/nvidia/jetson-inference/build/CMakeFiles/CMakeTmp'
Makefile:127: recipe for target 'cmTC_588eb/fast' failed
make: *** [cmTC_588eb/fast] Error 2

Source file was:
#include <pthread.h>

static void* test_func(void* data)
{
  return data;
}

int main(void)
{
  pthread_t thread;
  pthread_create(&thread, NULL, test_func, NULL);
  pthread_detach(thread);
  pthread_cancel(thread);
  pthread_join(thread, NULL);
  pthread_atfork(NULL, NULL, NULL);
  pthread_exit(NULL);

  return 0;
}

Determining if the function pthread_create exists in the pthreads failed with the following output:
Change Dir: /home/nvidia/jetson-inference/build/CMakeFiles/CMakeTmp

Run Build Command(s):/usr/bin/make -f Makefile cmTC_b82ea/fast && /usr/bin/make  -f CMakeFiles/cmTC_b82ea.dir/build.make CMakeFiles/cmTC_b82ea.dir/build
make[1]: Entering directory '/home/nvidia/jetson-inference/build/CMakeFiles/CMakeTmp'
Building C object CMakeFiles/cmTC_b82ea.dir/CheckFunctionExists.c.o
/usr/bin/cc   -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create -o CMakeFiles/cmTC_b82ea.dir/CheckFunctionExists.c.o -c /home/nvidia/cmake-3.20.4-linux-aarch64/share/cmake-3.20/Modules/CheckFunctionExists.c
Linking C executable cmTC_b82ea
/home/nvidia/cmake-3.20.4-linux-aarch64/bin/cmake -E cmake_link_script CMakeFiles/cmTC_b82ea.dir/link.txt --verbose=1
/usr/bin/cc -fPIC -DCHECK_FUNCTION_EXISTS=pthread_create -rdynamic CMakeFiles/cmTC_b82ea.dir/CheckFunctionExists.c.o -o cmTC_b82ea  -lpthreads 
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
CMakeFiles/cmTC_b82ea.dir/build.make:98: recipe for target 'cmTC_b82ea' failed
make[1]: *** [cmTC_b82ea] Error 1
make[1]: Leaving directory '/home/nvidia/jetson-inference/build/CMakeFiles/CMakeTmp'
Makefile:127: recipe for target 'cmTC_b82ea/fast' failed
make: *** [cmTC_b82ea/fast] Error 2

Is there anyone could help me, appreciate it if you leave me some instruction or guide.

ps:I have two Xaviers, succeed on Xavier I, failed on Xavier II. This is very weird that Xavier I and II both are jetpack 4.5 flashed by the same PC(Ubuntu 18.04) sdkmanager.

YaoJiawei329, 2021-09-03:
Emm, I found where is wrong. Nothing to do with Python3.7! When I use git clone to clone the repo, ../tools/camera-capture is empty, and I use $git submodule update --init, didn't work. Maybe in the EAST of the earth, Internet is not friendly with GITHUB.

HGGshiwo, 2022-05-06:
I have the same problem, 
```
/usr/bin/ld: cannot find -lpthreads
collect2: error: ld returned 1 exit status
```
And tried sudo apt-get install libpthread-stubs0-dev, but it didn't work.
Could you tell me the solution? Much apprecaite.

yyjqr, 2023-03-25:
I alse see this problem "python3.7 not found" when update the code.But the true error is container submodule files is not complete,some files can't find. SO NEED TO “git submodule update --init”  in container directory!

closed 2023-04-27 by dusty-nv