dusty-nv/jetson-inference #1206: Much less frame rate (FPS) than in the tutorial on Jetson Nano

sas9624, 2021-08-31:
The problems I encountered while running the inference part on Jetson Nano Developer Kit are

1. When I reproduced the task in the tutorial "[Locating Objects with DetectNet](https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-console-2.md#processing-video-files)".  By running `./detectnet.py /usr/share/visionworks/sources/data/pedestrians.mp4`, the frame rate of TensorRT is only around 10 FPS, while in the [video tutorial](https://www.youtube.com/watch?v=obt60r8ZeB0&list=PL5B692fm6--uQRRDTPsJDp4o0xbzkoyf8&index=12&ab_channel=NVIDIADeveloper), you got around 22 FPS.

2. I re-trained the mb1-ssd model by fruit dataset and ran the inference without TensorRT by following the upstream repo using [run_ssd_example.py](run_ssd_example.py) and then counted the number of frames and timed the inference time to find the average frames per second.  When compared with TensorRT which operating with 15FPS, TensorRT does not provide a distinctive improvement in terms of frame rate from running the inference on GPU with 12FPS (without TensorRT). 

I have done this experiment with Jetson Xavier earlier and TensorRT provided a distinctive improvement from running without GPU. To be specific the frame rate was around 100FPS with TensorRT and 30FPS without TensorRT. 

Do you have any ideas what can be the problems with TensorRT on Jeyson Nano?

These are some details about the Jetson Nano I am using:

> 
> PyTorch version: 1.8.0
> Is debug build: False
> CUDA used to build PyTorch: 10.2
> ROCM used to build PyTorch: N/A
> 
> OS: Ubuntu 18.04.5 LTS (aarch64)
> GCC version: (Ubuntu/Linaro 7.5.0-3ubuntu1~18.04) 7.5.0
> Clang version: Could not collect
> CMake version: version 3.10.2
> 
> Python version: 3.6 (64-bit runtime)
> Is CUDA available: True
> CUDA runtime version: Could not collect
> GPU models and configuration: Could not collect
> Nvidia driver version: Could not collect
> cuDNN version: Probably one of the following:
> /usr/lib/aarch64-linux-gnu/libcudnn.so.8.0.0
> /usr/lib/aarch64-linux-gnu/libcudnn_adv_infer.so.8.0.0
> /usr/lib/aarch64-linux-gnu/libcudnn_adv_train.so.8.0.0
> /usr/lib/aarch64-linux-gnu/libcudnn_cnn_infer.so.8.0.0
> /usr/lib/aarch64-linux-gnu/libcudnn_cnn_train.so.8.0.0
> /usr/lib/aarch64-linux-gnu/libcudnn_ops_infer.so.8.0.0
> /usr/lib/aarch64-linux-gnu/libcudnn_ops_train.so.8.0.0
> HIP runtime version: N/A
> MIOpen runtime version: N/A
> 
> Versions of relevant libraries:
> [pip3] numpy==1.19.5
> [pip3] torch==1.8.0
> [pip3] torchvision==0.9.0
> [conda] Could not collect

dusty-nv, 2021-08-31:
Hi @sas9624, is your Nano running in 5W mode or 10W mode?

You can check with `sudo nvpmodel -q`

sas9624, 2021-09-01:
Hello @dusty-nv, thank you for your reply.
The results from` sudo nvpmodel -q` are

> NVPM WARN: fan mode is not set!
> NV Power Mode: MAXN
> 0

So I think it is 10W mode.

May I also ask? Is it because Nano has no Tensor core, inference with TensorRT and without TensorRT(only on GPU) make not a big difference?

dusty-nv, 2021-09-01:
Hmm that is quite odd.  Can you delete the .engine file under `jetson-inference/data/networks/SSD-Mobilenet-v2` and try running detectnet on the pedestrians.mp4 again?

sas9624, 2021-09-01:
Hello @dusty-nv , I have done that but the FPS is still at around 7-8 FPS. 

I have used detectnet.py and then add this line `print(f"FPS is {net.GetNetworkFPS()}")` to print the frame rate
as I run it on RTP and it doesn't show the frame rate.

Do you have any suggestions?

dusty-nv, 2021-09-01:
Do you have a web browser or other applications running?  If so, can you try closing them.

Normally if this were a power supply issue, the board would just shut down if the board wasn't getting enough power.

Can you try running `sudo tegrastats` and paste some of the tegrastats output here from when detectnet is running?

sas9624, 2021-09-01:
Thank you for your fast response. 
There was nothing else running, I also tried rebooting it first before running detectnet but it made no difference.

Here are some results while running detectnet: 

> RAM 1501/3956MB (lfb 332x4MB) IRAM 0/252kB(lfb 252kB) CPU [16%@1479,13%@1479,10%@1479,4%@1479] EMC_FREQ 4%@1600 GR3D_FREQ 28%@76 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39C CPU@44C PMIC@100C GPU@41.5C AO@47.5C thermal@42.5C POM_5V_IN 4122/3107 POM_5V_GPU 109/85 POM_5V_CPU 1167/748
> RAM 1504/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [44%@1479,22%@1479,26%@1479,14%@1479] EMC_FREQ 7%@1600 GR3D_FREQ 86%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39C CPU@44C PMIC@100C GPU@42.5C AO@47.5C thermal@42.5C POM_5V_IN 4531/3285 POM_5V_GPU 1113/213 POM_5V_CPU 539/722
> RAM 1504/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [19%@1479,27%@1479,23%@1479,19%@1479] EMC_FREQ 9%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39.5C CPU@43.5C PMIC@100C GPU@43C AO@48C thermal@43C POM_5V_IN 4523/3422 POM_5V_GPU 1113/313 POM_5V_CPU 538/701
> RAM 1504/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [29%@1479,36%@1479,18%@1479,11%@1479] EMC_FREQ 10%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39.5C CPU@43.5C PMIC@100C GPU@43C AO@48C thermal@43.25C POM_5V_IN 4531/3533 POM_5V_GPU 1256/407 POM_5V_CPU 466/678
> RAM 1504/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [21%@825,39%@825,22%@825,20%@825] EMC_FREQ 11%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39C CPU@43.5C PMIC@100C GPU@42.5C AO@48C thermal@43C POM_5V_IN 4531/3624 POM_5V_GPU 1220/481 POM_5V_CPU 502/662
> RAM 1504/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [26%@1479,26%@1479,25%@1479,23%@1479] EMC_FREQ 11%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@40C CPU@43.5C PMIC@100C GPU@43C AO@48C thermal@43C POM_5V_IN 4531/3699 POM_5V_GPU 1148/537 POM_5V_CPU 502/648
> RAM 1504/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [33%@1224,19%@1224,21%@1224,30%@1224] EMC_FREQ 12%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39.5C CPU@43.5C PMIC@100C GPU@43C AO@48C thermal@43.25C POM_5V_IN 4523/3763 POM_5V_GPU 1184/587 POM_5V_CPU 466/634
> RAM 1505/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [23%@614,13%@614,21%@614,36%@614] EMC_FREQ 12%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39.5C CPU@43.5C PMIC@100C GPU@43C AO@48C thermal@43.25C POM_5V_IN 4496/3815 POM_5V_GPU 1294/637 POM_5V_CPU 395/617
> RAM 1505/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [39%@921,27%@921,17%@921,25%@921] EMC_FREQ 12%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39.5C CPU@44C PMIC@100C GPU@43C AO@48C thermal@43.25C POM_5V_IN 4523/3862 POM_5V_GPU 1256/678 POM_5V_CPU 466/607
> RAM 1505/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [20%@710,30%@710,33%@710,19%@710] EMC_FREQ 12%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@40C CPU@44C PMIC@100C GPU@43C AO@48C thermal@43.25C POM_5V_IN 4531/3904 POM_5V_GPU 1256/714 POM_5V_CPU 430/596
> RAM 1505/3956MB (lfb 331x4MB) IRAM 0/252kB(lfb 252kB) CPU [31%@710,26%@710,24%@710,20%@710] EMC_FREQ 12%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@40C CPU@44C PMIC@100C GPU@43C AO@48C thermal@43.5C POM_5V_IN 4496/3939 POM_5V_GPU 1292/748 POM_5V_CPU 395/584
> RAM 1505/3956MB (lfb 330x4MB) IRAM 0/252kB(lfb 252kB) CPU [24%@921,20%@921,27%@921,23%@921] EMC_FREQ 12%@1600 GR3D_FREQ 99%@921 NVDEC 716 VIC_FREQ 0%@140 APE 25 PLL@39.5C CPU@44C PMIC@100C GPU@43C AO@48C thermal@43.5C POM_5V_IN 4496/3970 POM_5V_GPU 1222/775 POM_5V_CPU 431/576

dusty-nv, 2021-09-01:
Hmm ok.  What power supply are you using?

sas9624, 2021-09-02:
Hello @dusty-nv, we use the developer kit's power supply.

dusty-nv, 2021-09-02:
> Hello @dusty-nv, we use the developer kit's power supply.

The Jetson Nano Developer Kit does not come with it's own power supply, you need to provide one.  Which one are you using?

sas9624, 2021-09-03:
Sorry for that, we are using the normal mobile phone power supply. 

I have just found your post about power supply
https://forums.developer.nvidia.com/t/power-supply-considerations-for-jetson-nano-developer-kit/71637
Which one do you recommend in order to get the same performance as in the tutorial?

closed 2023-04-27 by dusty-nv