dusty-nv/jetson-inference #1211: Gray image input:poseNet works not well

Wo-Ki, 2021-09-02:
Hi, when I test the poseNet, there is a good result for RGB image input, but bad for Gray. I try to adjust the threshold, just a little keypoints output, even the body in the image is full.
I think I need help, thank you very much.

dusty-nv, 2021-09-02:
Hi Wo-Ki, the models are for color images, so it is not surprising they don't work on grayscale.  I think it is meant for color camera/video input.

Wo-Ki, 2021-09-03:
This seems a good question. I have tested keypoints demo from [Detectron2](https://github.com/facebookresearch/detectron2),the model was trained with COCO datasets, which are almostly RGB images, but have a good performance on Gray input image. I think the model may not only learn color information ,but also texture.
I doubt the image preprocess, for example the decoding way, is this question?