dusty-nv/jetson-inference #1214: jetson-inference build in docker container on jetson NX / jetpack 4.6

asimonov, 2021-09-05:
@dusty-nv , thank you for the great work!

I can build and run jetson-inference on intel based docker.
And i can build manually inside a running docker on jetson.

But right now struggling to build jetson-inference inside docker on jetson/jetpack 4.6 at build time.

The problem: jetson docker mounts files from outside at runtime.

For example, at build time jetson-inference does not see tensorrt headers/libs.

I solved this by 

```
RUN echo "deb https://repo.download.nvidia.com/jetson/common r32.6 main" >> /etc/apt/sources.list && \
    apt update && \
    apt install -y tensorrt
```

but how it fails on some dependencies that go to `nvidia-l4t-core` (e.g. /usr/lib/aarch64-linux-gnu/tegra/libnvdla_compiler.so)

if I try this

```
RUN echo "deb https://repo.download.nvidia.com/jetson/t194 r32.6 main" >> /etc/apt/sources.list && \
    apt update && \
    apt install -y nvidia-l4t-core
```

then at docker build time I see:

```
Preparing to unpack .../nvidia-l4t-core_32.6.1-20210726122859_arm64.deb ...
/var/lib/dpkg/tmp.ci/preinst: line 40: /proc/device-tree/compatible: No such file or directory
dpkg: error processing archive /var/cache/apt/archives/nvidia-l4t-core_32.6.1-20210726122859_arm64.deb (--unpack):
 new nvidia-l4t-core package pre-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 /var/cache/apt/archives/nvidia-l4t-core_32.6.1-20210726122859_arm64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```
Looks like dependency on devices which are probably not in docker at build time...

PS
My base image is docker.io/arm64v8/ubuntu:18.04, I do not rely on nvidia images because i want everything spelled out in dockerfiles, not in some binary image.

What is the best way to have all the dependencies for jetson-inference while building a docker on jetpack 4.6?

asimonov, 2021-09-05:
or alternatively, is there a way to build it outside docker and map inside it using the same mechanism cuda/cudnn etc is mapped on jetson?

dusty-nv, 2021-09-06:
Hi Alexey, you need to set your default docker-runtime to nvidia:   https://github.com/dusty-nv/jetson-containers#docker-default-runtime

Then those libs will get mounted in.  I only support building this container on Jetson at this time.

asimonov, 2021-09-06:
Oh, lovely! This solves the build problem!

closed 2021-09-06 by asimonov