dusty-nv/jetson-inference #1216: how can I combine road following and object detection?

yangrisheng, 2021-09-07:
how can I combine road following and object detection?

dusty-nv, 2021-09-07:
Hi @yangrisheng, this project doesn't do road following - maybe you meant to ask this question on the [JetBot repo](https://github.com/nvidia-ai-iot/jetbot)?

yangrisheng, 2021-09-08:
> 嗨@yangrisheng，这个项目不做道路跟踪 - 也许你是想在[JetBot repo](https://github.com/nvidia-ai-iot/jetbot)上问这个问题？
> 
> 否则，您只需将对象检测代码添加到调用道路跟随的代码中，从而形成管道。

yes,yes.I use jetbot.I want to combine road following and object detection.But in jetbot project only has using object detection with a current engine.I want to combine them with my own dataset.I do not finish it.
this one is a problem. 
https://forums.developer.nvidia.com/t/in-jetson-nano-using-my-own-mobilenetv2-model/188671

closed 2023-04-27 by dusty-nv