dusty-nv/jetson-inference #1217: Adapting YOLOX- where can I do custom postprocessing?

tensorturtle, 2021-09-08:
Hello, I'm seeking your help in adapting [YOLOX](https://github.com/Megvii-BaseDetection/YOLOX) models to the `jetson-inference` platform.

So far I have succeded in using Python `tensorrt` and `pycuda` libraries to run the yolox-nano model at around 10fps on the Jetson Nano, with the neural network inference step taking around 0.05 seconds.  YOLOX-nano appears to have much better accuracy and small object detection capability than the provided Mobilenet-SSD model.

The problem is, my Python-based multiprocessing pipeline appears to have a lot of overhead, so I want to use the faster code in this repository.

The yolox-nano network outputs a tensor of size (batch, num_anchor_boxes (typically 3549 for a 416x416pixel input) , 85).

where 85 = 4 (bounding box coordinates)  + 1 (objectness score) +  80 (confidences for each COCO class)

I'm assuming that this output is formatted differently from the provided DetectNet outputs. Could you please point me to which code I need to change in order to do the required postprocessing?

Thank you for the work in writing and maintaining this codebase.

tensorturtle, 2021-09-08:
[Link to example postprocessing script for YOLOX](https://github.com/Megvii-BaseDetection/YOLOX/blob/main/demo/ONNXRuntime/onnx_inference.py)

dusty-nv, 2021-09-08:
Hi @tensorturtle, here is the pre- and post-processing code that you would need to adapt:

* https://github.com/dusty-nv/jetson-inference/blob/99387c24627e71fd93b2451d4b22e7de2e261be2/c/detectNet.cpp#L713
* https://github.com/dusty-nv/jetson-inference/blob/99387c24627e71fd93b2451d4b22e7de2e261be2/c/detectNet.cpp#L800

tensorturtle, 2021-09-08:
Thank you, much appreciated.

tensorturtle, 2021-09-09:
Hello @dusty-nv , I have been studying the pre- and post-processing code in `detectNet.cpp`. I am new to C++, and I would appreciate more help in understanding the code.

With regards to Line 800~ :

Does the provided Mobilenet-SSD model output two tensors (arrays) as its output?
 
Am I correct in understanding that these arrays are saved in `mOutputs`, in index 0 and 1?
 
What is `mOutputs`, exactly?

Is `clusterDetections` an implementation of Non Maximum Suppression? 
How is it different from Non Maximum Suppression?

dusty-nv, 2021-09-09:
> Does the provided Mobilenet-SSD model output two tensors (arrays) as its output?
> 
> Am I correct in understanding that these arrays are saved in `mOutputs`, in index 0 and 1?

That is correct.  One tensor is a grid of confidences and the other is a grid of bounding box offsets.

> What is `mOutputs`, exactly?

It's an array of output tensors from TensorRT.  See this struct for the type:

https://github.com/dusty-nv/jetson-inference/blob/99387c24627e71fd93b2451d4b22e7de2e261be2/c/tensorNet.h#L740

> Is `clusterDetections` an implementation of Non Maximum Suppression?
> How is it different from Non Maximum Suppression?

Essentially, although I probably took some shortcuts trying to make it faster.

tensorturtle, 2021-09-10:
Thank you for the detailed guidance.

tensorturtle, 2021-09-13:
> Hi @tensorturtle, here is the pre- and post-processing code that you would need to adapt:
>
>     * [...]

Hello @dusty-nv ,
These places in code are where I should place pre- and post-processing code for ONNX models, right?

What if I already have a custom TensorRT `.engine` file? Should pre- and post-processing code go in the `IsModelType ( MODEL_UFF )` sections? Is UFF the same as TensoRT Engine serialized file?