dusty-nv/jetson-inference #1218: Unknown error 999,help me!

liumarcus70s, 2021-09-10:
Hi all gays,
    I got a nano 2GB and then download the Jetson Nano 2GB Developer Kit SD Card Image from the official website.
    I flash the image to an SD card, run the nano and clone the inference from Github.
    Then there comes the problem.
    After cloning the repository, I used docker to run the install application firstly. But when I run the "camera-viewer" or "camera-viewer.py", errors came.
    Then I compiled the codes, still the same errors as following.
===================================================================================
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/marcus2/codes/jetson-inference/utils/display/glTexture.cpp:360
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/marcus2/codes/jetson-inference/utils/cuda/cudaYUV-YV12.cu:119
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/marcus2/codes/jetson-inference/utils/cuda/cudaColorspace.cpp:59
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/marcus2/codes/jetson-inference/utils/camera/gstCamera.cpp:728
[gstreamer] gstCamera::Capture() -- unsupported image format (rgba32f)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f
camera-viewer:  failed to capture RGBA image
=================================================================================
The camera I used was a Logitech C922 pro which can work on my 1st gen jetson nano 4GB.
So, any idea to deal with?

liumarcus70s, 2021-10-09:
Hi @dusty-nv, Thanks for your help.
Unfortunately, I can't reappear the error again after reflashing the system img.

NipaDasGupta, 2021-12-27:
I got the same error in my Nvidia Jetson Nano 2GB Developer Kit.

nano@linux:~/jetson-inference$ docker/run.sh
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.6.1
[sudo] password for nano: 
size of data/networks:  623468514 bytes
CONTAINER:     dustynv/jetson-inference:r32.6.1
DATA_VOLUME:   --volume /home/nano/jetson-inference/data:/jetson-inference/data --volume /home/nano/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/nano/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/nano/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/nano/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
root@linux:/jetson-inference# ls /dev/video*
/dev/video0
root@linux:/jetson-inference# video-viewer /dev/video0
nvbuf_utils: Could not get EGL display connection
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: Rapoo Camera
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Rapoo\ Camera", v4l2.device.bus_info=(string)usb-70090000.xusb-3.2, v4l2.device.version=(uint)264701, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 11 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)10/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)20/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [5] image/jpeg, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 3/1 };
[gstreamer] [6] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [7] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [8] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [9] image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [10] image/jpeg, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] gstCamera -- selected device profile:  codec=mjpeg format=unknown width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 ! image/jpeg, width=(int)1280, height=(int)720 ! jpegdec ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      mjpeg
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
[... 538 lines cut ...]
video-viewer:  captured 32 frames (1280 x 720)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/display/glTexture.cpp:360
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/cuda/cudaYUV-YV12.cu:119
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/cuda/cudaColorspace.cpp:53
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/camera/gstCamera.cpp:728
[gstreamer] gstCamera::Capture() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f
video-viewer:  failed to capture video frame
video-viewer:  captured 33 frames (1280 x 720)
[OpenGL] glDisplay -- the window has been closed
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/display/glTexture.cpp:360
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
video-viewer:  shutdown complete
root@linux:/jetson-inference#

NipaDasGupta, 2021-12-27:
nano@linux:~/jetson-inference$ docker/run.sh
reading L4T version from /etc/nv_tegra_release
L4T BSP Version:  L4T R32.6.1
[sudo] password for nano: 
size of data/networks:  623468514 bytes
CONTAINER:     dustynv/jetson-inference:r32.6.1
DATA_VOLUME:   --volume /home/nano/jetson-inference/data:/jetson-inference/data --volume /home/nano/jetson-inference/python/training/classification/data:/jetson-inference/python/training/classification/data --volume /home/nano/jetson-inference/python/training/classification/models:/jetson-inference/python/training/classification/models --volume /home/nano/jetson-inference/python/training/detection/ssd/data:/jetson-inference/python/training/detection/ssd/data --volume /home/nano/jetson-inference/python/training/detection/ssd/models:/jetson-inference/python/training/detection/ssd/models
USER_VOLUME:   
USER_COMMAND:  
V4L2_DEVICES:    --device /dev/video0 
localuser:root being added to access control list
root@linux:/jetson-inference# video-viewer /dev/video0
nvbuf_utils: Could not get EGL display connection
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: Rapoo Camera
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Rapoo\ Camera", v4l2.device.bus_info=(string)usb-70090000.xusb-3.2, v4l2.device.version=(uint)264701, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 11 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)5/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)10/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)20/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [5] image/jpeg, width=(int)2560, height=(int)1440, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 25/1, 3/1 };
[gstreamer] [6] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [7] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [8] image/jpeg, width=(int)800, height=(int)600, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [9] image/jpeg, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [10] image/jpeg, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] gstCamera -- selected device profile:  codec=mjpeg format=unknown width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 ! image/jpeg, width=(int)1280, height=(int)720 ! jpegdec ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      mjpeg
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1920x1080
[OpenGL] glDisplay -- X window resolution:    1920x1080
[OpenGL] glDisplay -- display device initialized (1920x1080)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1920
  -- height:     1080
  -- frameRate:  0.000000
  -- bitRate:    0
[... 54 lines cut ...]
video-viewer:  captured 2 frames (1280 x 720)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/display/glTexture.cpp:360
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/cuda/cudaYUV-YV12.cu:119
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/cuda/cudaColorspace.cpp:53
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/camera/gstCamera.cpp:728
[gstreamer] gstCamera::Capture() -- unsupported image format (rgb8)
[gstreamer]                         supported formats are:
[gstreamer]                             * rgb8
[gstreamer]                             * rgba8
[gstreamer]                             * rgb32f
[gstreamer]                             * rgba32f
video-viewer:  failed to capture video frame
video-viewer:  captured 21 frames (1280 x 720)
[OpenGL] glDisplay -- the window has been closed
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /jetson-inference/utils/display/glTexture.cpp:360
video-viewer:  shutting down...
[gstreamer] gstCamera -- stopping pipeline, transitioning to GST_STATE_NULL
[gstreamer] gstCamera -- pipeline stopped
video-viewer:  shutdown complete
root@linux:/jetson-inference#

tonylau2010, 2022-01-18:
Hi！ @dusty-nv 
I got the same error in my  Jetson Nano 4GB kit.Would you help me,thanks very much.

The log as follows：
tony-nano@tonynano-desktop:~/Downloads/readyZip/jetson-inference/build/aarch64/bin$ video-viewer /dev/video0**
nvbuf_utils: Could not get EGL display connection
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: USB2.0 Camera
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"USB2.0\ Camera", v4l2.device.bus_info=(string)usb-70090000.xusb-2.2, v4l2.device.version=(uint)264701, v4l2.device.capabilities=(uint)2216689665, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 5 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] gstCamera -- selected device profile:  codec=raw format=yuyv width=640 height=480
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 ! video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480 ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[video]  created gstCamera from v4l2:///dev/video0
------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      raw
  -- width:      640
  -- height:     480
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1364x768
[OpenGL] glDisplay -- X window resolution:    1364x768
[OpenGL] glDisplay -- display device initialized (1364x768)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1364
  -- height:     768
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[gstreamer] opening gstCamera for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter0
[gstreamer] gstreamer changed state from NULL to READY ==> v4l2src0
[gstreamer] gstreamer changed state from NULL to READY ==> pipeline0
[gstreamer] gstreamer changed state from READY to PAUSED ==> capsfilter0
[gstreamer] gstreamer stream status CREATE ==> src
[... 27 lines cut ...]
[gstreamer] gstBufferManager -- unsupported image format (rgb8)
[gstreamer]                     supported formats are:
[gstreamer]                        * rgb8
[gstreamer]                        * rgba8
[gstreamer]                        * rgb32f
[gstreamer]                        * rgba32f
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame
video-viewer:  captured 2 frames (640 x 480)
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/display/glTexture.cpp:360
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/cuda/cudaYUV-YUYV.cu:160
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/cuda/cudaColorspace.cpp:75
[cuda]      unknown error (error 999) (hex 0x3E7)
[cuda]      /home/tony-nano/Downloads/readyZip/jetson-inference/utils/codec/gstBufferManager.cpp:383
[gstreamer] gstBufferManager -- unsupported image format (rgb8)
[gstreamer]                     supported formats are:
[gstreamer]                        * rgb8
[gstreamer]                        * rgba8
[gstreamer]                        * rgb32f
[gstreamer]                        * rgba32f
[gstreamer] gstDecoder -- failed to retrieve next image buffer
video-viewer:  failed to capture video frame

dusty-nv, 2022-01-18:
Hi @tonylau2010, it seems there's an error processing your USB camera.   As a test, does it work for you if you play video files like this?  https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-console-2.md#processing-video-files

Which camera are you using?  Do you happen to have another camera you can try?

tonylau2010, 2022-01-18:
> Hi @tonylau2010, it seems there's an error processing your USB camera. As a test, does it work for you if you play video files like this? [...]

Thanks! @dusty-nv I will try it.  Now one usb camera is using,Iwill try another one.

By the way,I use command to show  local mp4 video,another err is happen.
[gstreamer] gstBufferManager -- failed to map EGLImage from NVMM buffer
[gstreamer] gstDecoder -- failed to handle incoming buffer

---------full log is as follows:
tony-nano@tonynano-desktop:~/Downloads/readyZip/jetson-inference/build/aarch64/bin$ video-viewer cuda-course.mp4 
nvbuf_utils: Could not get EGL display connection
[gstreamer] initialized gstreamer, version 1.14.5.0
[gstreamer] gstDecoder -- creating decoder for cuda-course.mp4
Opening in BLOCKING MODE 
NvMMLiteOpen : Block : BlockType = 261 
NVMEDIA: Reading vendor.tegra.display-size : status: 6 
NvMMLiteBlockCreate : Block : BlockType = 261 
[gstreamer] gstDecoder -- discovered video resolution: 1920x1080  (framerate 15.000000 Hz)
[gstreamer] gstDecoder -- discovered video caps:  video/x-h264, stream-format=(string)byte-stream, alignment=(string)au, level=(string)4, profile=(string)constrained-baseline, width=(int)1920, height=(int)1080, framerate=(fraction)15/1, pixel-aspect-ratio=(fraction)1/1, interlace-mode=(string)progressive, chroma-format=(string)4:2:0, bit-depth-luma=(uint)8, bit-depth-chroma=(uint)8, parsed=(boolean)true
[gstreamer] gstDecoder -- pipeline string:
[gstreamer] filesrc location=cuda-course.mp4 ! qtdemux ! queue ! h264parse ! omxh264dec ! video/x-raw(memory:NVMM) ! appsink name=mysink
[video]  created gstDecoder from file:///home/tony-nano/Downloads/readyZip/jetson-inference/build/aarch64/bin/cuda-course.mp4
------------------------------------------------
gstDecoder video options:
------------------------------------------------
  -- URI: file:///home/tony-nano/Downloads/readyZip/jetson-inference/build/aarch64/bin/cuda-course.mp4
     - protocol:  file
     - location:  cuda-course.mp4
     - extension: mp4
  -- deviceType: file
  -- ioType:     input
  -- codec:      h264
  -- width:      1920
  -- height:     1080
  -- frameRate:  15.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  1364x768
[OpenGL] glDisplay -- X window resolution:    1364x768
[OpenGL] glDisplay -- display device initialized (1364x768)
[video]  created glDisplay from display://0
------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      1364
  -- height:     768
  -- frameRate:  0.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[gstreamer] opening gstDecoder for streaming, transitioning pipeline to GST_STATE_PLAYING
[gstreamer] gstreamer changed state from NULL to READY ==> mysink
[gstreamer] gstreamer changed state from NULL to READY ==> capsfilter1
[gstreamer] gstreamer changed state from NULL to READY ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from NULL to READY ==> h264parse1
[... 35 lines cut ...]
[gstreamer] gstreamer message duration-changed ==> h264parse1
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ /\ AVC", language-code=(string)en, bitrate=(uint)378535;
[gstreamer] gstreamer mysink taglist, datetime=(datetime)2022-01-12T00:58:29Z, container-format=(string)"ISO\ MP4/M4A";
[gstreamer] gstreamer mysink taglist, video-codec=(string)"H.264\ \(Constrained\ Baseline\ Profile\)", language-code=(string)en, bitrate=(uint)378535;
[gstreamer] gstBufferManager -- map buffer size was less than max size (1008 vs 3110400)
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), format=(string)NV12, width=(int)1920, height=(int)1080, interlace-mode=(string)progressive, multiview-mode=(string)mono, multiview-flags=(GstVideoMultiviewFlagsSet)0:ffffffff:/right-view-first/left-flipped/left-flopped/right-flipped/right-flopped/half-aspect/mixed-mono, pixel-aspect-ratio=(fraction)1/1, chroma-site=(string)mpeg2, colorimetry=(string)bt709, framerate=(fraction)15/1
[gstreamer] gstBufferManager -- recieved first frame, codec=h264 format=nv12 width=1920 height=1080 size=3110400
[gstreamer] gstBufferManager -- recieved NVMM memory
NvEGLImageFromFd: No EGLDisplay to create EGLImage
[gstreamer] gstBufferManager -- failed to map EGLImage from NVMM buffer
[gstreamer] gstDecoder -- failed to handle incoming buffer
[gstreamer] gstreamer changed state from READY to PAUSED ==> mysink
[gstreamer] gstreamer changed state from READY to PAUSED ==> pipeline0
[gstreamer] gstreamer message async-done ==> pipeline0
[gstreamer] gstreamer message new-clock ==> pipeline0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> mysink
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> capsfilter1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> omxh264dec-omxh264dec0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> h264parse1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> queue0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> qtdemux1
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> filesrc0
[gstreamer] gstreamer changed state from PAUSED to PLAYING ==> pipeline0
[gstreamer] gstBufferManager -- map buffer size was less than max size (1008 vs 3110400)
[gstreamer] gstBufferManager recieve caps:  video/x-raw(memory:NVMM), format=(string)NV12

oriaj3, 2022-10-25:
Hi, I have the same bug, I access the jetson via xrdp and I get the 999 error both with an rtsp video stream from an ip camera, and with an mp4 following these steps. https://github.com/dusty-nv/jetson-inference/blob/master/docs/detectnet-console-2.md#processing-video-files

franz6ko, 2022-11-01:
I'm having the same problem. I'm trying to build jeston_inference on my laptop.

I cloned the nvmm-disabled repo and added my cuda 11.1 path and the architecture(sm_86) on the CMakeList.txt. Then build but when I run any demo or even the video-viewer with my C920 usb camera I get this.

[image]

I also had to change the omxh264dec step for avdec_h264 in gstCamera.cpp and gstDecoder.cpp

tonylau2010, 2022-11-01:
you can try run script directly,not with xrdp

franz6ko, 2022-11-01:
I'm not using any remote desktop tool if that's what you're suggesting.

I tried a lot of things but I'm not getting any results. Using the docker image without doing any changes to the repo (master branch) gives a Segmentation Fault. Using the docker image on the dev branch gives the same cuda errors shown above.

It's weird as I'm not compiling anything now (using docker images) and I get the same memory-related problems.

franz6ko, 2022-11-01:
Ok, I figured it out !! Hope this will help anybody in the same situation.

Apparently, the problem was with the memory reservation on glTexture.cpp:360. I did some research and OpenGL context was being created on my non-NVIDIA GPU (I have both an integrated GPU and a dedicated GPU on my laptop). See [this](https://forums.developer.nvidia.com/t/cudaerrorunknown-cudagraphicsglregisterbuffer/64406).

What I did to solve it is to run "sudo prime-select nvidia" so that everything uses the Nvidia GPU. Maybe the best solution would be to tell explicitely to OpenGL to run on the Nvidia GPU but I don't know how to do that.

Hope this helps somebody, I spent two days breaking my brain trying to solve this.

ryuchanhoe, 2022-11-30:
Hi @franz6ko , sorry for keep pulling you out. 
I am trying the jetson inference at my laptop, x86 with build in cam without using docker container.
I believe I am having the same problem as you and "sudo prime-select nvidia" is not the remedy for me.
I confirmed ./segnet is working with sample images, and my cam working with ffplay /dev/video0.
I was wondering if you had to do anything else than "sudo prime-select nvidia"?
You mentioned "I also had to change the omxh264dec step for avdec_h264 in gstCamera.cpp and gstDecoder.cpp"
Is this still a thing? 
If yes, can you please explain more about that? Thank you.

command line: video-viewer /dev/video0
[image]

[gstreamer] initialized gstreamer, version 1.16.3.0
[gstreamer] gstCamera -- attempting to create device v4l2:///dev/video0
[gstreamer] gstCamera -- found v4l2 device: Integrated_Webcam_FHD: Integrat
[gstreamer] v4l2-proplist, device.path=(string)/dev/video0, udev-probed=(boolean)false, device.api=(string)v4l2, v4l2.device.driver=(string)uvcvideo, v4l2.device.card=(string)"Integrated_Webcam_FHD:\ Integrat", v4l2.device.bus_info=(string)usb-0000:00:14.0-9, v4l2.device.version=(uint)331584, v4l2.device.capabilities=(uint)2225078273, v4l2.device.device_caps=(uint)69206017;
[gstreamer] gstCamera -- found 12 caps for v4l2 device /dev/video0
[gstreamer] [0] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 30/1 };
[gstreamer] [1] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 30/1 };
[gstreamer] [2] video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [3] video/x-raw, format=(string)YUY2, width=(int)424, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [4] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [5] video/x-raw, format=(string)YUY2, width=(int)320, height=(int)180, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [6] video/x-raw, format=(string)YUY2, width=(int)160, height=(int)120, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [7] image/jpeg, width=(int)1920, height=(int)1080, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [8] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 30/1 };
[gstreamer] [9] image/jpeg, width=(int)1280, height=(int)720, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction){ 30/1, 30/1 };
[gstreamer] [10] image/jpeg, width=(int)960, height=(int)540, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] [11] image/jpeg, width=(int)848, height=(int)480, pixel-aspect-ratio=(fraction)1/1, framerate=(fraction)30/1;
[gstreamer] gstCamera -- selected device profile:  codec=mjpeg format=unknown width=1280 height=720
[gstreamer] gstCamera pipeline string:
[gstreamer] v4l2src device=/dev/video0 do-timestamp=true ! image/jpeg, width=(int)1280, height=(int)720 ! jpegdec ! video/x-raw ! appsink name=mysink
[gstreamer] gstCamera successfully created device v4l2:///dev/video0
[0;32m[video]  created gstCamera from v4l2:///dev/video0
[0m------------------------------------------------
gstCamera video options:
------------------------------------------------
  -- URI: v4l2:///dev/video0
     - protocol:  v4l2
     - location:  /dev/video0
  -- deviceType: v4l2
  -- ioType:     input
  -- codec:      mjpeg
  -- width:      1280
  -- height:     720
  -- frameRate:  30.000000
  -- bitRate:    0
  -- numBuffers: 4
  -- zeroCopy:   true
  -- flipMethod: none
  -- loop:       0
  -- rtspLatency 2000
------------------------------------------------
[OpenGL] glDisplay -- X screen 0 resolution:  3840x2160
[OpenGL] glDisplay -- X window resolution:    3840x2160
[OpenGL] glDisplay -- display device initialized (3840x2160)
[0;32m[video]  created glDisplay from display://0
[0m------------------------------------------------
glDisplay video options:
------------------------------------------------
  -- URI: display://0
     - protocol:  display
     - location:  0
  -- deviceType: display
  -- ioType:     output
  -- codec:      raw
  -- width:      3840
  -- height:     2160
  -- frameRate:  0.000000
  -- bitRate:    0
[... 88 lines cut ...]
[0m[0;31m[gstreamer]                        * rgb8
[0m[0;31m[gstreamer]                        * rgba8
[0m[0;31m[gstreamer]                        * rgb32f
[0m[0;31m[gstreamer]                        * rgba32f
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mvideo-viewer:  failed to capture video frame
[0mvideo-viewer:  captured 4 frames (1280 x 720)
[0;31m[cuda]      unknown error (error 999) (hex 0x3E7)
[0m[0;31m[cuda]      /home/usrg/jetson-inference/utils/display/glTexture.cpp:360
[0m[0;31m[cuda]      unknown error (error 999) (hex 0x3E7)
[0m[0;31m[cuda]      /home/usrg/jetson-inference/utils/cuda/cudaYUV-YV12.cu:119
[0m[0;31m[cuda]      unknown error (error 999) (hex 0x3E7)
[0m[0;31m[cuda]      /home/usrg/jetson-inference/utils/cuda/cudaColorspace.cpp:53
[0m[0;31m[cuda]      unknown error (error 999) (hex 0x3E7)
[0m[0;31m[cuda]      /home/usrg/jetson-inference/utils/codec/gstBufferManager.cpp:435
[... cut ...]
[0m[0;31m[gstreamer]                     supported formats are:
[0m[0;31m[gstreamer]                        * rgb8
[0m[0;31m[gstreamer]                        * rgba8
[0m[0;31m[gstreamer]                        * rgb32f
[0m[0;31m[gstreamer]                        * rgba32f
[0m[0;31m[gstreamer] gstDecoder -- failed to retrieve next image buffer
[0m[0;31mvideo-viewer:  failed to capture video frame
[0mvideo-viewer:  captured 5 frames (1280 x 720)
[0;31m[cuda]      unknown error (error 999) (hex 0x3E7)

closed 2023-04-27 by dusty-nv

LIM0422, 2025-07-30:
xhost +local:root
export DISPLAY=:0
try this,may solve it