dusty-nv/jetson-inference #1220: TRTassertion failed for Detecnet with MobileNetv1/v2 and Inception

fduxvtech, 2021-09-13:
Hi, TRT  cannot create the engine given the default parameters in detecnet:
`Create("networks/SSD-Mobilenet-v1/ssd_mobilenet_v1_coco.uff", "networks/SSD-Mobilenet-v1/ssd_coco_labels.txt", threshold, "Input", Dims3(3,300,300), "Postprocessor", "Postprocessor_1", maxBatchSize, precision, device, allowGPUFallback)`

The initial output seems ok:

> detectNet -- loading detection network model from:
          -- model        networks/SSD-Mobilenet-v1/ssd_mobilenet_v1_coco.uff
          -- input_blob   'Input'
          -- output_blob  'Postprocessor'
          -- output_count 'Postprocessor_1'
          -- class_labels networks/SSD-Mobilenet-v1/ssd_coco_labels.txt
          -- threshold    0.500000
          -- batch_size   1
  [TRT]    TensorRT version 7.1.3
  [TRT]    loading NVIDIA plugins...
  [TRT]    Registered plugin creator - ::GridAnchor_TRT version 1
  [TRT]    Registered plugin creator - ::NMS_TRT version 1
  [TRT]    Registered plugin creator - ::Reorg_TRT version 1
  [TRT]    Registered plugin creator - ::Region_TRT version 1
  [TRT]    Registered plugin creator - ::Clip_TRT version 1
  [TRT]    Registered plugin creator - ::LReLU_TRT version 1
  [TRT]    Registered plugin creator - ::PriorBox_TRT version 1
  [TRT]    Registered plugin creator - ::Normalize_TRT version 1
  [TRT]    Registered plugin creator - ::RPROI_TRT version 1
  [TRT]    Registered plugin creator - ::BatchedNMS_TRT version 1
  [TRT]    Registered plugin creator - ::FlattenConcat_TRT version 1
  [TRT]    Registered plugin creator - ::CropAndResize version 1
  [TRT]    Registered plugin creator - ::DetectionLayer_TRT version 1
  [TRT]    Registered plugin creator - ::Proposal version 1
  [TRT]    Registered plugin creator - ::ProposalLayer_TRT version 1
  [TRT]    Registered plugin creator - ::PyramidROIAlign_TRT version 1
  [TRT]    Registered plugin creator - ::ResizeNearest_TRT version 1
  [TRT]    Registered plugin creator - ::Split version 1
  [TRT]    Registered plugin creator - ::SpecialSlice_TRT version 1
  [TRT]    Registered plugin creator - ::InstanceNormalization_TRT version 1
  [TRT]    detected model format - UFF  (extension '.uff')
  [TRT]    desired precision specified for GPU: FASTEST
  [TRT]    requested fasted precision for device GPU without providing valid calibrator, disabling INT8
  [TRT]    native precisions detected for GPU:  FP32, FP16, INT8
  [TRT]    selecting fastest native precision for GPU:  FP16
  [TRT]    attempting to open engine cache file /usr/local/bin/networks/SSD-Mobilenet-v1/ssd_mobilenet_v1_coco.uff.1.1.7103.GPU.FP16.engine
  [TRT]    cache file not found, profiling network model on device GPU

After this the UFFParser start and the error is:
>[TRT]    UFFParser: Parsing concat_box_loc[Op: _FlattenConcat_TRT]. Inputs: BoxPredictor_0/Reshape, BoxPredictor_1/Reshape, BoxPredictor_2/Reshape, BoxPredictor_3/Reshape, BoxPredictor_4/Reshape, BoxPredictor_5/Reshape
#assertionflattenConcat.cpp,49

Any idea on the cause of this? 
Thanks

dusty-nv, 2021-10-04:
Hmm.  Does this work with the default SSD-Mobilenet-v2 model, or just get the error with SSD-Mobilenet-v1?

Also, which Jetson are you using and which version of JetPack?

fduxvtech, 2021-10-05:
Hey dusty, it does happens in my application; if I run the detectnet example it doesn't happens; it should be a problem related to the conversion on [Aasta part](https://forums.developer.nvidia.com/t/problem-with-ssd-mobilenet-tensorrt-conversion/161284/8)

fduxvtech, 2021-10-05:
The main difference is that in my application I use jetson-inference as sub-module; in my app it does find and registrer the FlattenConcat_TRT plugin and I have the assertion error.
If I run the example detecnet from the installed version of jetson-inference then it doesn't find the plugin and remain unregistered.

Following my edit in the previous comment:
> EDIT: it is jetpack v4.4 and I was able to install tensorflow 1.15, I'll try the solution proposed in the forum.
I'm not sure why but the uffparser in python is totally broken with my tensorflow version.

fduxvtech, 2021-10-14:
@dusty-nv any ideas?

closed 2023-04-27 by dusty-nv