dusty-nv/jetson-inference #1221: Collecting your own Detection Datasets automatically

eljuligallego, 2021-09-13:
Hi,
just curious about if could be possible to automatically collect the objects for the datasets instead of take 100 pictures where each one I have to adjust the rounding box manually and save it.

I mean, should be easy to paint a box around some object different from background and assign the label you decide for this set of pictures automatically.

Maybe anyone have done it, Just don't want to reinvent the wheel.

dusty-nv, 2021-09-13:
Hi @eljuligallego I believe the CVAT tool has some vision-assisted labeling features, although I haven't used them.  CVAT is what I use on images that I already have.  It can export to Pascal VOC format.

eljuligallego, 2021-09-13:
found this:
https://github.com/SkalskiP/make-sense

has online version and automatically labeling and export to VOC too but not sure if 100% compatible.

closed 2021-09-13 by eljuligallego