dusty-nv/jetson-inference #1223: A question regarding an idea

neilyoung, 2021-09-16:
Hi Dusty, 

Maybe you remember me: We had some nice conversations in the beginning of this year and I always admired your prompt and founded replies. That time you pointed me to the DeepStream SDK (for performance), and I finally was able to bring a 3 USB camera object detection app for the Jetson Nano to life, which showed an incredible inference rate of 30 fps per camera...

Anyway, a new project is "ante portas", for which I would like to ask you for your opinion and maybe some guidance. 

This time I have to go to the cloud with AI. The video is transported to the cloud entity using WebRTC and terminated by a Media Server solution with SFU capabilities. One outstanding characteristic of the Media Server is: It allows you to create OpenCV plugins, which give you access to any single frame sent by remote. Basically it all lands in an OpenCV::Mat.

I'm not sure if the DeepStream SDK allows for a plain "frame input" (I always just dealt with GStreamer pipelines there), but I suppose, your code does support that. The question now is: Is your code able to be run apart from the Jetson Nano, e.g. on a cloud computer, which has an Nvidia GPU somewhere in the back? 

TIA
Regards

neilyoung, 2021-09-17:
@dusty-nv Any pointer?

dusty-nv, 2021-09-17:
Hi @neilyoung, in the past some folks have managed to compile this project for x86, but I don't officially support it.  You would also need to change out the Jetson-specific gstreamer elements for the standard ones that are used on PC.

If you are doing hybrid cloud/edge, then DeepStream is well-positioned for that, because it can run on both Jetson and x86, and has message brokers integrated.

dusty-nv, 2021-09-17:
> The guys you are mentioning - did they appear hear in this repo?

You may be able to find some mentions in old/closed GitHub issues on this repo

neilyoung, 2021-09-17:
*here :)

Thanks

closed 2023-04-27 by dusty-nv