dusty-nv/jetson-inference #1225: OSError: [Errno 12] Cannot allocate memory

SurawutSukkum, 2021-09-17:
I just try to understand the object Detection (Re-training SSD MobileNet). When i record dataset, don't have the problem.
but when i re-training with Re-training SSD MobileNet  found the problem is OSError: [Errno 12] Cannot allocate memory.
How to solve this problem?

surawut@surawut-desktop:~/jetson-inference/python/training/detection/ssd$ python3 train_ssd.py --dataset-type=voc --data=data/pinacle --model-dir=models/pinacle --batch-size=2 --workers=1 --epochs=1
2021-09-17 17:53:22 - Using CUDA...
2021-09-17 17:53:22 - Namespace(balance_data=False, base_net=None, base_net_lr=0.001, batch_size=2, checkpoint_folder='models/pinacle', dataset_type='voc', datasets=['data/pinacle'], debug_steps=10, extra_layers_lr=None, freeze_base_net=False, freeze_net=False, gamma=0.1, lr=0.01, mb2_width_mult=1.0, milestones='80,100', momentum=0.9, net='mb1-ssd', num_epochs=1, num_workers=1, pretrained_ssd='models/mobilenet-v1-ssd-mp-0_675.pth', resume=None, scheduler='cosine', t_max=100, use_cuda=True, validation_epochs=1, weight_decay=0.0005)
2021-09-17 17:53:22 - Prepare training datasets.
warning - image 20210917-165231 has no box/labels annotations, ignoring from dataset
2021-09-17 17:53:22 - VOC Labels read from file: ('BACKGROUND', 'tap', 'Battery', '', '', '', '')
2021-09-17 17:53:22 - Stored labels into file models/pinacle/labels.txt.
2021-09-17 17:53:22 - Train dataset size: 1067
2021-09-17 17:53:22 - Prepare Validation datasets.
2021-09-17 17:53:23 - VOC Labels read from file: ('BACKGROUND', 'tap', 'Battery', '', '', '', '')
2021-09-17 17:53:23 - Validation dataset size: 925
2021-09-17 17:53:23 - Build network.
2021-09-17 17:53:23 - Init from pretrained ssd models/mobilenet-v1-ssd-mp-0_675.pth
2021-09-17 17:53:24 - Took 0.56 seconds to load the model.
2021-09-17 17:53:37 - Learning rate: 0.01, Base net learning rate: 0.001, Extra Layers learning rate: 0.01.
2021-09-17 17:53:37 - Uses CosineAnnealingLR scheduler.
2021-09-17 17:53:37 - Start training from epoch 0.
/home/surawut/.local/lib/python3.6/site-packages/torch/optim/lr_scheduler.py:123: UserWarning: Detected call of `lr_scheduler.step()` before `optimizer.step()`. In PyTorch 1.1.0 and later, you should call them in the opposite order: `optimizer.step()` before `lr_scheduler.step()`.  Failure to do this will result in PyTorch skipping the first value of the learning rate schedule. See more details at https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate
  "https://pytorch.org/docs/stable/optim.html#how-to-adjust-learning-rate", UserWarning)
/home/surawut/.local/lib/python3.6/site-packages/torch/nn/_reduction.py:44: UserWarning: size_average and reduce args will be deprecated, please use reduction='sum' instead.
  warnings.warn(warning.format(ret))
2021-09-17 17:54:47 - Epoch: 0, Step: 10/534, Avg Loss: 26.1412, Avg Regression Loss 12.0325, Avg Classification Loss: 14.1087
2021-09-17 17:54:51 - Epoch: 0, Step: 20/534, Avg Loss: 18.1214, Avg Regression Loss 8.3359, Avg Classification Loss: 9.7855
2021-09-17 17:54:57 - Epoch: 0, Step: 30/534, Avg Loss: 13.1593, Avg Regression Loss 7.8744, Avg Classification Loss: 5.2849
2021-09-17 17:55:01 - Epoch: 0, Step: 40/534, Avg Loss: 9.6254, Avg Regression Loss 5.1580, Avg Classification Loss: 4.4674
2021-09-17 17:55:06 - Epoch: 0, Step: 50/534, Avg Loss: 11.2497, Avg Regression Loss 6.0238, Avg Classification Loss: 5.2259
2021-09-17 17:55:15 - Epoch: 0, Step: 60/534, Avg Loss: 11.7554, Avg Regression Loss 7.2765, Avg Classification Loss: 4.4789
2021-09-17 17:55:20 - Epoch: 0, Step: 70/534, Avg Loss: 9.7318, Avg Regression Loss 5.2363, Avg Classification Loss: 4.4955
2021-09-17 17:55:25 - Epoch: 0, Step: 80/534, Avg Loss: 9.0316, Avg Regression Loss 4.8262, Avg Classification Loss: 4.2054
2021-09-17 17:55:39 - Epoch: 0, Step: 90/534, Avg Loss: 10.4938, Avg Regression Loss 5.9147, Avg Classification Loss: 4.5791
2021-09-17 17:55:55 - Epoch: 0, Step: 100/534, Avg Loss: 9.1556, Avg Regression Loss 4.6432, Avg Classification Loss: 4.5123
2021-09-17 17:56:00 - Epoch: 0, Step: 110/534, Avg Loss: 10.3318, Avg Regression Loss 5.5270, Avg Classification Loss: 4.8048
2021-09-17 17:56:09 - Epoch: 0, Step: 120/534, Avg Loss: 7.5285, Avg Regression Loss 3.4963, Avg Classification Loss: 4.0323
2021-09-17 17:56:15 - Epoch: 0, Step: 130/534, Avg Loss: 8.2938, Avg Regression Loss 4.6879, Avg Classification Loss: 3.6059
2021-09-17 17:56:24 - Epoch: 0, Step: 140/534, Avg Loss: 9.2254, Avg Regression Loss 5.2273, Avg Classification Loss: 3.9981
2021-09-17 17:56:33 - Epoch: 0, Step: 150/534, Avg Loss: 7.5854, Avg Regression Loss 4.1966, Avg Classification Loss: 3.3889
2021-09-17 17:56:39 - Epoch: 0, Step: 160/534, Avg Loss: 8.0956, Avg Regression Loss 4.3235, Avg Classification Loss: 3.7721
2021-09-17 17:56:46 - Epoch: 0, Step: 170/534, Avg Loss: 7.6432, Avg Regression Loss 4.0700, Avg Classification Loss: 3.5733
2021-09-17 17:56:53 - Epoch: 0, Step: 180/534, Avg Loss: 7.0740, Avg Regression Loss 3.7920, Avg Classification Loss: 3.2820
2021-09-17 17:56:58 - Epoch: 0, Step: 190/534, Avg Loss: 8.1043, Avg Regression Loss 4.7182, Avg Classification Loss: 3.3860
2021-09-17 17:57:08 - Epoch: 0, Step: 200/534, Avg Loss: 7.8231, Avg Regression Loss 4.2308, Avg Classification Loss: 3.5923
2021-09-17 17:57:14 - Epoch: 0, Step: 210/534, Avg Loss: 8.3173, Avg Regression Loss 4.8184, Avg Classification Loss: 3.4988
2021-09-17 17:57:21 - Epoch: 0, Step: 220/534, Avg Loss: 7.1365, Avg Regression Loss 3.8329, Avg Classification Loss: 3.3036
2021-09-17 17:57:28 - Epoch: 0, Step: 230/534, Avg Loss: 7.0919, Avg Regression Loss 3.4566, Avg Classification Loss: 3.6353
2021-09-17 17:57:33 - Epoch: 0, Step: 240/534, Avg Loss: 5.7697, Avg Regression Loss 2.5840, Avg Classification Loss: 3.1857
2021-09-17 17:57:39 - Epoch: 0, Step: 250/534, Avg Loss: 6.8505, Avg Regression Loss 3.6152, Avg Classification Loss: 3.2353
2021-09-17 17:57:46 - Epoch: 0, Step: 260/534, Avg Loss: 8.8895, Avg Regression Loss 5.2640, Avg Classification Loss: 3.6255
2021-09-17 17:57:52 - Epoch: 0, Step: 270/534, Avg Loss: 7.7015, Avg Regression Loss 4.0353, Avg Classification Loss: 3.6662
2021-09-17 17:57:56 - Epoch: 0, Step: 280/534, Avg Loss: 6.7228, Avg Regression Loss 3.2366, Avg Classification Loss: 3.4862
2021-09-17 17:58:06 - Epoch: 0, Step: 290/534, Avg Loss: 6.6907, Avg Regression Loss 3.1546, Avg Classification Loss: 3.5361
2021-09-17 17:58:13 - Epoch: 0, Step: 300/534, Avg Loss: 7.4413, Avg Regression Loss 3.9700, Avg Classification Loss: 3.4714
2021-09-17 17:58:23 - Epoch: 0, Step: 310/534, Avg Loss: 6.5453, Avg Regression Loss 3.0076, Avg Classification Loss: 3.5376
2021-09-17 17:58:37 - Epoch: 0, Step: 320/534, Avg Loss: 7.7675, Avg Regression Loss 4.2840, Avg Classification Loss: 3.4835
2021-09-17 17:58:42 - Epoch: 0, Step: 330/534, Avg Loss: 6.0360, Avg Regression Loss 2.9606, Avg Classification Loss: 3.0755
2021-09-17 17:58:47 - Epoch: 0, Step: 340/534, Avg Loss: 6.1408, Avg Regression Loss 2.9225, Avg Classification Loss: 3.2183
2021-09-17 17:58:54 - Epoch: 0, Step: 350/534, Avg Loss: 6.8534, Avg Regression Loss 3.4765, Avg Classification Loss: 3.3769
2021-09-17 17:58:58 - Epoch: 0, Step: 360/534, Avg Loss: 5.5556, Avg Regression Loss 2.6379, Avg Classification Loss: 2.9177
2021-09-17 17:59:02 - Epoch: 0, Step: 370/534, Avg Loss: 5.1561, Avg Regression Loss 2.1724, Avg Classification Loss: 2.9837
2021-09-17 17:59:07 - Epoch: 0, Step: 380/534, Avg Loss: 5.1102, Avg Regression Loss 2.3442, Avg Classification Loss: 2.7660
2021-09-17 17:59:11 - Epoch: 0, Step: 390/534, Avg Loss: 5.6748, Avg Regression Loss 2.6424, Avg Classification Loss: 3.0324
2021-09-17 17:59:16 - Epoch: 0, Step: 400/534, Avg Loss: 5.7002, Avg Regression Loss 2.5393, Avg Classification Loss: 3.1609
2021-09-17 17:59:27 - Epoch: 0, Step: 410/534, Avg Loss: 6.9364, Avg Regression Loss 3.7954, Avg Classification Loss: 3.1410
2021-09-17 17:59:32 - Epoch: 0, Step: 420/534, Avg Loss: 6.1661, Avg Regression Loss 2.5494, Avg Classification Loss: 3.6167
2021-09-17 17:59:36 - Epoch: 0, Step: 430/534, Avg Loss: 4.6479, Avg Regression Loss 1.8474, Avg Classification Loss: 2.8005
2021-09-17 17:59:41 - Epoch: 0, Step: 440/534, Avg Loss: 4.9370, Avg Regression Loss 2.1042, Avg Classification Loss: 2.8328
2021-09-17 17:59:46 - Epoch: 0, Step: 450/534, Avg Loss: 5.3147, Avg Regression Loss 2.3749, Avg Classification Loss: 2.9398
[... 3 lines cut ...]
[... cut ...]
  File "/usr/lib/python3.6/multiprocessing/context.py", line 223, in _Popen
    return _default_context.get_context().Process._Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/context.py", line 277, in _Popen
    return Popen(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 19, in __init__
    self._launch(process_obj)
  File "/usr/lib/python3.6/multiprocessing/popen_fork.py", line 66, in _launch
    self.pid = os.fork()
OSError: [Errno 12] Cannot allocate memory

dusty-nv, 2021-09-17:
Hi @SurawutSukkum, try running train_ssd.py with `--batch-size=1 --workers=0`

dusty-nv, 2021-09-17:
Also try these suggestions if you haven't already:

* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
* https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

SurawutSukkum, 2021-09-18:
> Also try these suggestions if you haven't already:
> 
> * https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
> * https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

I was disable my screen( $ sudo init 3  ) and then my screen can't recovery to enable it. I suddenly turn off power and then turn on it. My screen always dark.

dusty-nv, 2021-09-18:
Press ctrl + alt + 1  (or 2 or 3, I can't remember) and it should bring up login prompt

  *   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
  *   https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

SurawutSukkum, 2021-09-21:
> Also try these suggestions if you haven't already:
> 
> * https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#mounting-swap
> * https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-transfer-learning.md#disabling-the-desktop-gui

python3 train_ssd.py --dataset-type=voc --data=data/pinacle --model-dir=models/pinacle --batch-size=1 --workers=0 --epochs=1

I did it and  use change --batch-size=1 --workers=0. The result is work. Thank you very much.

closed 2021-09-21 by SurawutSukkum