dusty-nv/jetson-inference #1228: Instead of using python2.7, jetson fails build trying to use python3.8

Emmanuel-Messulam, 2021-09-20:
```cmake
-- detecting Python 2.7...
-- found Python version:  2.7 (2.7.18)
-- found Python include:  /usr/include/python2.7
-- found Python library:  /usr/lib/x86_64-linux-gnu/libpython2.7.so
-- CMake module path:  /home/emmanuelm/Documentos/test/jetson-inference/utils/cuda;/home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings
-- NumPy ver. 1.16.5 found (include: /usr/lib/python2.7/dist-packages/numpy/core/include)
-- found NumPy version:  1.16.5
-- found NumPy include:  /usr/lib/python2.7/dist-packages/numpy/core/include
-- detecting Python 3.6...
-- found Python version:  3.8 (3.8.10)
-- found Python include:  /usr/include/python3.8
-- found Python library:  /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- CMake module path:  /home/emmanuelm/Documentos/test/jetson-inference/utils/cuda;/home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings
-- NumPy ver. 1.17.4 found (include: /usr/lib/python3/dist-packages/numpy/core/include)
-- found NumPy version:  1.17.4
-- found NumPy include:  /usr/lib/python3/dist-packages/numpy/core/include
-- detecting Python 3.7...
-- found Python version:  3.8 (3.8.10)
-- found Python include:  /usr/include/python3.8
-- found Python library:  /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- CMake module path:  /home/emmanuelm/Documentos/test/jetson-inference/utils/cuda;/home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings
-- NumPy ver. 1.17.4 found (include: /usr/lib/python3/dist-packages/numpy/core/include)
-- found NumPy version:  1.17.4
-- found NumPy include:  /usr/lib/python3/dist-packages/numpy/core/include
CMake Error at utils/cuda/FindCUDA.cmake:1802 (add_library):
  add_library cannot create target "jetson-utils-python-38" because another
  target with the same name already exists.  The existing target is a shared
  library created in source directory
  "/home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  utils/python/bindings/CMakeLists.txt:57 (cuda_add_library)

CMake Error at utils/python/bindings/CMakeLists.txt:59 (target_link_libraries):
  Attempt to add link library "jetson-utils" to target
  "jetson-utils-python-38" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

CMake Error at utils/python/bindings/CMakeLists.txt:62 (target_link_libraries):
  Attempt to add link library "npymath" to target "jetson-utils-python-38"
  which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.
```
[CMakeOutput.log](https://github.com/dusty-nv/jetson-inference/files/7196211/CMakeOutput.log)

It should use python2.7 instead of using python3.8

dusty-nv, 2021-09-20:
@Emmanuel-Messulam if you run `python3 --version` does it report python 3.6 or python 3.8?

I think you need to not have python3 sym'linked to python3.8

Emmanuel-Messulam, 2021-09-20:
Python 3.8.10

> I think you need to not have python3 sym'linked to python3.8

Doing that might break other things in my setup.

dusty-nv, 2021-09-20:
OK.  Try removing everything but the python version you want from these lines:

* https://github.com/dusty-nv/jetson-inference/blob/cb732028d79b26ab909a6fce3020894c13e21689/python/CMakeLists.txt#L8
* https://github.com/dusty-nv/jetson-utils/blob/1ee0494b196b488b4b95e89fa2d72366d3cf4879/python/CMakeLists.txt#L8

So if you just want Python 3.8, only put Python 3.8 there.

Emmanuel-Messulam, 2021-09-20:
> OK. Try removing everything but the python version you want from these lines:
> 
>     * https://github.com/dusty-nv/jetson-inference/blob/cb732028d79b26ab909a6fce3020894c13e21689/python/CMakeLists.txt#L8
> 
>     * [dusty-nv/jetson-utils@`1ee0494`/python/CMakeLists.txt#L8](https://github.com/dusty-nv/jetson-utils/blob/1ee0494b196b488b4b95e89fa2d72366d3cf4879/python/CMakeLists.txt#L8)
> 
> 
> So if you just want Python 3.8, only put Python 3.8 there.

I want python 2.7, but it seems to be grabbing 3.8, I already tried setting /python/CMakeLists.txt to `set(PYTHON_BINDING_VERSIONS 2.7)`.

dusty-nv, 2021-09-20:
From your log above, it detected 2.7 correctly but detected 3.6 and 3.8.

Try either temporarily restoring the original python/python3 symlinks when you build the project, or build the container.

Emmanuel-Messulam, 2021-09-20:
I'll try to open an issue to fix the build problem with python 3.8

closed 2021-09-20 by Emmanuel-Messulam