dusty-nv/jetson-inference #1229: Python 3.8 build issue

Emmanuel-Messulam, 2021-09-20:
Trying to build using python 3.8 I having an issue with CUDA.

```cmake
-- trying to build Python bindings for Python versions:  3.8
-- detecting Python 3.8...
-- found Python version:  3.8 (3.8.10)
-- found Python include:  /usr/include/python3.8
-- found Python library:  /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- CMake module path:  /home/emmanuelm/Documentos/test/jetson-inference/utils/cuda;/home/emmanuelm/Documentos/test/jetson-inference/python/bindings;/home/emmanuelm/Documentos/test/jetson-inference/python/bindings/../../utils/python/bindings
-- NumPy ver. 1.17.4 found (include: /usr/lib/python3/dist-packages/numpy/core/include)
-- found NumPy version:  1.17.4
-- found NumPy include:  /usr/lib/python3/dist-packages/numpy/core/include
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/depthnet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/depthnet_utils.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/detectnet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/imagenet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/my-detection.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/my-recognition.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/posenet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/segnet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/segnet_utils.py
-- Copying examples/imagenet.py -> imagenet-console.py
-- Copying examples/imagenet.py -> imagenet-camera.py
-- Copying examples/detectnet.py -> detectnet-console.py
-- Copying examples/detectnet.py -> detectnet-camera.py
-- Copying examples/segnet.py -> segnet-console.py
-- Copying examples/segnet.py -> segnet-camera.py
CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
Please set them or make sure they are set and tested correctly in the CMake files:
CUDA_CUDART_LIBRARY (ADVANCED)
    linked by target "jetson-inference" in directory /home/emmanuelm/Documentos/test/jetson-inference
    linked by target "imagenet-console" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/imagenet
    linked by target "imagenet-camera" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/imagenet
    linked by target "imagenet" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/imagenet
    linked by target "depthnet" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/depthnet
    linked by target "detectnet-console" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/detectnet
    linked by target "detectnet-camera" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/detectnet
    linked by target "detectnet" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/detectnet
    linked by target "posenet" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/posenet
    linked by target "segnet-console" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/segnet
    linked by target "segnet-camera" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/segnet
    linked by target "segnet" in directory /home/emmanuelm/Documentos/test/jetson-inference/examples/segnet
    linked by target "camera-capture" in directory /home/emmanuelm/Documentos/test/jetson-inference/tools/camera-capture
    linked by target "jetson-utils" in directory /home/emmanuelm/Documentos/test/jetson-inference/utils
    linked by target "gl-display-test" in directory /home/emmanuelm/Documentos/test/jetson-inference/utils/display/gl-display-test
    linked by target "jetson-utils-python-27" in directory /home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings
    linked by target "jetson-utils-python-38" in directory /home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings
    linked by target "jetson-inference-python-38" in directory /home/emmanuelm/Documentos/test/jetson-inference/python/bindings
CUDA_TOOLKIT_INCLUDE (ADVANCED)
   used as include directory in directory /home/emmanuelm/Documentos/test/jetson-inference

```

[CMakeOutput.log](https://github.com/dusty-nv/jetson-inference/files/7197991/CMakeOutput.log)

I have changed to `set(PYTHON_BINDING_VERSIONS 3.8)` in python/CMakeLists.txt.

dusty-nv, 2021-09-20:
> `/usr/lib/x86_64-linux-gnu/libpython3.8.so`

You are building this on x86, but this project is only officially supported on Jetson.  There are a number of Jetson-specific things, mostly in the jetson-utils module (such as GStreamer pipelines, ect)

Do you have CUDA Toolkit installed on your PC under `/usr/local/cuda`?   Is `/usr/local/cuda/bin` in your `$PATH`?

Emmanuel-Messulam, 2021-09-20:
> > `/usr/lib/x86_64-linux-gnu/libpython3.8.so`
> 
> You are building this on x86, but this project is only officially supported on Jetson. There are a number of Jetson-specific things, mostly in the jetson-utils module (such as GStreamer pipelines, ect)

> Do you have CUDA Toolkit installed on your PC under `/usr/local/cuda`? 

Yes, I have now symlinked `/usr/local/nvidia-cuda-toolkit`, to `/usr/local/cuda`.

> Is `/usr/local/cuda/bin` in your `$PATH`?

No, I have `/usr/local/nvidia-cuda-toolkit/bin`.
An error persists:

```
-- found Python version:  3.8 (3.8.10)
-- found Python include:  /usr/include/python3.8
-- found Python library:  /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- CMake module path:  /home/emmanuelm/Documentos/test/jetson-inference/utils/cuda;/home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings
-- NumPy ver. 1.17.4 found (include: /usr/lib/python3/dist-packages/numpy/core/include)
-- found NumPy version:  1.17.4
-- found NumPy include:  /usr/lib/python3/dist-packages/numpy/core/include
CMake Error at utils/cuda/FindCUDA.cmake:1802 (add_library):
  add_library cannot create target "jetson-utils-python-38" because another
  target with the same name already exists.  The existing target is a shared
  library created in source directory
  "/home/emmanuelm/Documentos/test/jetson-inference/utils/python/bindings".
  See documentation for policy CMP0002 for more details.
Call Stack (most recent call first):
  utils/python/bindings/CMakeLists.txt:57 (cuda_add_library)

CMake Error at utils/python/bindings/CMakeLists.txt:59 (target_link_libraries):
  Attempt to add link library "jetson-utils" to target
  "jetson-utils-python-38" which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

CMake Error at utils/python/bindings/CMakeLists.txt:62 (target_link_libraries):
  Attempt to add link library "npymath" to target "jetson-utils-python-38"
  which is not built in this directory.

  This is allowed only when policy CMP0079 is set to NEW.

-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/camera-viewer.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/cuda-examples.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/cuda-from-cv.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/cuda-from-numpy.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/cuda-to-cv.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/cuda-to-numpy.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/gl-display-test.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/utils/python/examples/video-viewer.py
-- trying to build Python bindings for Python versions:  3.8
-- detecting Python 3.8...
-- found Python version:  3.8 (3.8.10)
-- found Python include:  /usr/include/python3.8
-- found Python library:  /usr/lib/x86_64-linux-gnu/libpython3.8.so
-- CMake module path:  /home/emmanuelm/Documentos/test/jetson-inference/utils/cuda;/home/emmanuelm/Documentos/test/jetson-inference/python/bindings;/home/emmanuelm/Documentos/test/jetson-inference/python/bindings/../../utils/python/bindings
-- NumPy ver. 1.17.4 found (include: /usr/lib/python3/dist-packages/numpy/core/include)
-- found NumPy version:  1.17.4
-- found NumPy include:  /usr/lib/python3/dist-packages/numpy/core/include
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/depthnet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/depthnet_utils.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/detectnet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/imagenet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/my-detection.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/my-recognition.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/posenet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/segnet.py
-- Copying /home/emmanuelm/Documentos/test/jetson-inference/python/examples/segnet_utils.py
-- Copying examples/imagenet.py -> imagenet-console.py
-- Copying examples/imagenet.py -> imagenet-camera.py
-- Copying examples/detectnet.py -> detectnet-console.py
-- Copying examples/detectnet.py -> detectnet-camera.py
-- Copying examples/segnet.py -> segnet-console.py
-- Copying examples/segnet.py -> segnet-camera.py
-- Configuring incomplete, errors occurred!
See also "/home/emmanuelm/Documentos/test/jetson-inference/build/CMakeFiles/CMakeOutput.log".

```

vigneshoff, 2022-04-04:
Hi, 

Did you try to resolve this issue, I am facing same exact issue.
And also I found there should some modification required as per the below thread

[https://github.com/OE4T/meta-tegra/issues/440]

If you want I can share the recipe which I am using.

Thanks
Vignesh

andresgalaviz, 2022-07-08:
I have the same issue when trying to build this inside the Jetpack 5.0 l4t-ml:r34.1.1-py3 container