dusty-nv/jetson-inference #1235: nvinfer1::IExecutionContext  cannot destroy()

guduzhixing, 2021-09-26:
I have NVIDIA Jetson Xavier NX(4.4.1),than I find some error in https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorNet.cpp : 1378
```
mContext = context;
```
this mContext can not destory in tensorNet::~tensorNet(),exampel my code is 
```
// Destructor
tensorNet::~tensorNet()
{
	if( mEngine != NULL )
	{
		mEngine->destroy();
		mEngine = NULL;
	}
		
	if( mInfer != NULL )
	{
		mInfer->destroy();
		mInfer = NULL;
	}

	if( mContext != NULL )
	{
		mContext->destroy();
		mmContextnfer = NULL;
	}
}
```
run code printf error info is 
Thread 1 "test_upedester_" received signal SIGSEGV, Segmentation fault.
0x0000007faa78e8e4 in nvinfer1::rt::SafeExecutionContext::~SafeExecutionContext() () from /usr/lib/aarch64-linux-gnu/libnvinfer.so.7
(gdb) bt
#0  0x0000007faa78e8e4 in nvinfer1::rt::SafeExecutionContext::~SafeExecutionContext() () at /usr/lib/aarch64-linux-gnu/libnvinfer.so.7
#1  0x0000007faa50eaa4 in nvinfer1::rt::ExecutionContext::~ExecutionContext() () at /usr/lib/aarch64-linux-gnu/libnvinfer.so.7
#2  0x0000007faa50eb94 in nvinfer1::rt::ExecutionContext::~ExecutionContext() () at /usr/lib/aarch64-linux-gnu/libnvinfer.so.7
#3  0x0000007fb7b0a71c in tensorNet::~tensorNet() (this=0x5556859980, __in_chrg=<optimized out>)

if not destory mContext,when I new and delete object,the memory while increasing.

But, I add code in  https://github.com/dusty-nv/jetson-inference/blob/master/c/tensorNet.cpp : 1434
```
context->destroy();
mContext = context;
```
this context success destory

harryxdl, 2021-09-27:
Hi,
You shoud destroy mContext before mEngine.

guduzhixing, 2021-09-27:
@harryxdl ,thanks,this is work!

closed 2021-09-27 by guduzhixing

dicky98, 2023-08-11:
what is destroy is marked deprecated now?

dusty-nv, 2023-08-11:
@dicky98 what version of JetPack/L4T or TensorRT are you building on?