dusty-nv/jetson-inference #1241: OpenCV include for detectnet.cpp

Nowfal1993, 2021-10-01:
I just wanted to include opencv to the `detectnet.cpp` as my input is of `yuv` and I wanted to convert it `rgb` and feed it into my `Detect()`. But when I treid to #include opencv modules, while make it shows `fatal error no such file or directory`. But in the root `cmakeList.txt`, it seems like opencv modules are linked. How can I tackle this. Thanks in Advance.

dusty-nv, 2021-10-01:
Hi @Nowfal1993, OpenCV only gets included in the CMakeLists.txt under certain conditions:

https://github.com/dusty-nv/jetson-inference/blob/21a2000b1a4dd8686bc89bd35cffecc19c1e562f/CMakeLists.txt#L103

I would move that line outside of the if statement if you always want to include/use OpenCV.  Or add that line into the CMakeLists.txt of the specific program/application that you are building.

Nowfal1993, 2021-10-04:
This works: add that line into the CMakeLists.txt of the specific program/application that you are building.
` find_package(OpenCV COMPONENTS core calib3d) `
Also added 
`target_link_libraries(detectnet jetson-inference opencv_core opencv_calib3d)`

closed 2021-10-04 by Nowfal1993