dusty-nv/jetson-inference #1242: Object detection demo does not work

Gianni86M, 2021-10-02:
Please help ASAP
I cannot go further because I have no idea why I downloaded human pictures from open images to perform the object detection demo and it says it cannot find ('/jetson-inference/data/images/Human_*.jpg')
i already completed the training and processed it to onnx

I followed every step in your video and this should not be happening

Gianni86M, 2021-10-02:
this is the complete error list
listDir('/jetson-inference/data/images/Human_*.jpg') - found no matches
[image]  imageLoader -- failed to find '/jetson-inference/data/images/Human_*.jpg'
detectnet:  failed to create input stream

Gianni86M, 2021-10-02:
I need to advance and finish everything that Nvidia has to offer

Gianni86M, 2021-10-03:
Can I please have an answer with due respect

Gianni86M, 2021-10-03:
?

dusty-nv, 2021-10-04:
You have typo, it's `"/jetson-inference/data/images/humans_*.jpg"`

closed 2023-04-27 by dusty-nv