dusty-nv/jetson-inference #1244: How to run MobilenetV2 image classification model on Nvidia Jetson (Not SSD_mobilenet)

jasvinderkhurana, 2021-10-03:
Hi

I have been trying hard to find how to run mobilenet model on Nvidia Jetson. In Imagenet the list of network supported is as below:

```
root@jasvinder-desktop:/jetson-inference/python/training/classification# imagenet --help
usage: imagenet [--help] [--network=NETWORK] ...
                input_URI [output_URI]

Classify a video/image stream using an image recognition DNN.
See below for additional arguments that may not be shown above.

positional arguments:
    input_URI       resource URI of input stream  (see videoSource below)
    output_URI      resource URI of output stream (see videoOutput below)

imageNet arguments: 
  --network=NETWORK    pre-trained model to load, one of the following:
                           * alexnet
                           * googlenet (default)
                           * googlenet-12
                           * resnet-18
                           * resnet-50
                           * resnet-101
                           * resnet-152
                           * vgg-16
                           * vgg-19
                           * inception-v4
  --model=MODEL        path to custom model to load (caffemodel, uff, or onnx)

```
I  don't see any mobilenet network here.

When I try to run my mobilenet.onnx model I have below error:

```
 ----- Parsing of ONNX model /jetson-inference/data/custom_inference//model.onnx is Done ---- 
[TRT]    device GPU, configuring network builder
[TRT]    device GPU, building FP16:  ON
[TRT]    device GPU, building INT8:  OFF
[TRT]    device GPU, workspace size: 33554432
[TRT]    device GPU, building CUDA engine (this may take a few minutes the first time a network is loaded)
[TRT]    Network has dynamic or shape inputs, but no optimization profile has been defined.
[TRT]    Network validation failed.
[TRT]    device GPU, failed to build CUDA engine
[TRT]    device GPU, failed to load /jetson-inference/data/custom_inference//model.onnx
[TRT]    failed to load /jetson-inference/data/custom_inference//model.onnx
[TRT]    imageNet -- failed to initialize.
jetson.inference -- imageNet failed to load built-in network 'googlenet'
Traceback (most recent call last):
  File "/usr/local/bin/imagenet.py", line 55, in <module>
    net = jetson.inference.imageNet(opt.network, sys.argv)
Exception: jetson.inference -- imageNet failed to load network

```

Can anybody please help

dusty-nv, 2021-10-04:
> Network has dynamic or shape inputs, but no optimization profile has been defined.

Hi @jasvinderkhurana, you need to export this model without dynamic input shapes.

jasvinderkhurana, 2021-10-04:
@dusty-nv , thanks a lot for your reply!  Can you please provide more details on:
1) How to export any model without dynamic input shapes
2) What is the reference application (C++ or python), to run inference on that model, because **imagenet** , sample application do not take Mobilenet network as input, check "**imagenet --help**"

'''
root@jasvinder-desktop:/jetson-inference/python/training/classification# imagenet --help
usage: imagenet [--help] [--network=NETWORK] ...
                input_URI [output_URI]

Classify a video/image stream using an image recognition DNN.
See below for additional arguments that may not be shown above.

positional arguments:
    input_URI       resource URI of input stream  (see videoSource below)
    output_URI      resource URI of output stream (see videoOutput below)

imageNet arguments: 
  --network=NETWORK    pre-trained model to load, one of the following:
                           * alexnet
                           * googlenet (default)
                           * googlenet-12
                           * resnet-18
                           * resnet-50
                           * resnet-101
                           * resnet-152
                           * vgg-16
                           * vgg-19
                           * inception-v4
  --model=MODEL        path to custom model to load (caffemodel, uff, or onnx)
'''

dusty-nv, 2021-10-04:
> 1. How to export any model without dynamic input shapes

What is your framework that you are exporting the ONNX model from?

This script I use with PyTorch to export classification models without using dynamic shapes:
https://github.com/dusty-nv/pytorch-classification/blob/393914b484fc946876e0a1fe37c971acdfb1fec6/onnx_export.py

jasvinderkhurana, 2021-10-04:
I am using tensorflow framework to get pretrained MobileNetV2 model, train for custom dataset and save checkpoint model (.pb), Freeze model graph and then finally use tf2onnx utility to convert to onnx. 

Used below command to save tf2 model:
```
tf.saved_model.save(model,'.\mobilenet_v2_1.0_224.tf2')

```

jasvinderkhurana, 2021-10-04:
I am not bound to any framework, but I am bound to train MobileNetV2 model with custom data set and run it on Nvidia Jetson Nano. I have also checked out below Pytorch examples for classification and Detection, and there is no reference for MobileNet model for image classification.

https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-cat-dog.md
https://github.com/dusty-nv/jetson-inference/blob/master/docs/pytorch-ssd.md

jasvinderkhurana, 2021-10-04:
Also BTW, I had already using pytorch references and faced issue with MobileNet there as well, raised below ticket for the same:
https://github.com/dusty-nv/pytorch-classification/issues/6